espressif/esp-idf #6: Guru Meditation Error

AdySan, 2016-09-03:
I just got my Demo board today and followed the steps in [this](https://github.com/espressif/esp-idf/blob/master/docs/macos-setup.rst) excellent readme.

This part regarding selecting serial port worked well.
[image]

`make flash` didn't work a couple of times, but then is suddenly worked and finished correctly

[image]

After that when I reboot the board and look at the serial trace, looks like the chip keep rebooting. Here's the trace 

```
+.@ 0.@ÿÿ?³..Initializing heap allocator:
Region 19: 3FFC0998 len 0001F668 tag 0
Region 25: 3FFE8000 len 00018000 tag 1
Pro cpu up.
Pro cpu start user code
erase sector 5
erase sector 6
erase sector 7
nvs_flash_init
misc_nvs_init g_misc_nvs=0x3ffc0b04
/Users/adityatannu/Developer/esp-idf/components/freertos/./tasks.c:601 (xTaskGenericCreate)- assert failed!
Guru Meditation Error: Core   0 panic'ed.
Register dump:
PC      : Guru Meditation Error of type LoadProhibited occured on core   0. Exception was unhandled.
Register dump:
PC      :  40083842  PS      :  00060033  A0      :  80083a65  A1      :  3ffe3a70  
A2      :  00000000  A3      :  00000002  A4      :  00000000  A5      :  00000000  
A6      :  3f40681c  A7      :  3ff4001c  A8      :  0000007d  A9      :  3ff4001c  
A10     :  3ff4001c  A11     :  3f400284  A12     :  00000000  A13     :  3ffe3ad0  
A14     :  00000000  A15     :  3f406930  SAR     :  0000001b  EXCCAUSE:  0000001c  
EXCVADDR:  00000004  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
Rebooting...
Ä¤§å¤bb¥äÄ&a..À¦çäÄ¤Ä.b..#.A#cÀbå§¦Åä."àÄbÅg¤åÄ¥Äd¤åÄ¦¥Ä¥Ä¥D.ä@.æå¦§bæ¦¥ÄçD.äÄ¥¥Æä§E.äÄ¤Æä§ÅF.äÄÄÆä¥bä¤ÄÄ¥bcä¥FCcf§ÅçççççççççççççççççççbcÄcä¥æÄÄæ¥¦..¥Gcf§Å¤¤ç¦¤äÇ§äbcÄcä¥æ¥æ&äÄGcf§ÅçççççççççççççççççççbcÄcä¥¦ÇÄÆ¥åÅbbcÄcä¥$@$BGcf§Å.âä.bbcÄcä¥$@$Å$æDAcf§ÅæÅäbcÄcä¥¦.À$£¥DäFcf§Å¤¤äÅ§bcÄcä¥¥$Dcf§Å¤¤¦ÁÁ¤bcÄcä¥.äæÅDcf§Åâææä¥bcÄcä¥¦¤ÄçÅÄ¤¤Å¥ÄÄ¤¤bcÄcä¥¦¤ÄçÅÄ¤¤¦Å¥Ä¤¥Dcf§Åæå¤ÄÄÅÄ¥¥¥¤æÄÄ¥ÅÄDcöì..ï§.âÆÇ§ÄÄæäåbÅçæ¤!Æ¡Å¤bÅç¤ä¤!À¤b G.å¤¤åb¤¤§b¤¤§¤b¤¤
```

Did I do something wrong?

AdySan, 2016-09-03:
FWIW, I did not rebuild the toolchain, just got the binaries from the espressif server.

Spritetm, 2016-09-03:
tasks.c:601 is configASSERT( (xCoreID>=0 && xCoreID<portNUM_PROCESSORS) || (xCoreID==tskNO_AFFINITY) );

it looks like you're trying to start a process on core 2 while FreeRTOS only has one core. Can you check in menuconfig if components->esp32->reserve memory for 2 cores is enabled, and components->freertos->run freertos only on first core is disabled?

By the way, 'guru meditation error' is basically a panic or unhandled exception.

AdySan, 2016-09-03:
Tried removing wifi as @igrr suggested on Twitter, and it works now

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:2312
load:0x40078000,len:3788
load:0x40098000,len:532
entry 0x4009813c
.[1;36m

.[0m.[1;36m**************************************
.[0m.[1;36m*       hello espressif ESP32!       *
.[0m.[1;36m*        2nd boot is running!        *
.[0m.[1;36m*            version (V0.1)          *
.[0m.[1;36m**************************************
.[0m.[0;36mcompile time 22:31:01

.[0m.[0;36m  SPI Speed      : 40MHz
.[0m.[0;36m  SPI Mode       : QIO
.[0m.[0;36m  SPI Flash Size : 4MB
.[0m.[1;36mPartition Table:
.[0m.[1;36m## Label            Usage          Type ST Offset   Length
.[0m.[1;36m 0 factory          factory app      00 00 00010000 00100000
.[0m.[1;36m 1 rfdata           RF data          01 01 00110000 00040000
.[0m.[1;36m 2 wifidata         WiFi data        01 02 00150000 00040000
.[0m.[1;36mEnd of partition table
.[0m.[1;36mLoading app partition at offset 00010000
.[0m.[0;36msection 0: paddr=0x00000020 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
.[0m.[0;36msection 1: paddr=0x00010010 vaddr=0x3f400010 size=0x00d48 (  3400) map
.[0m.[0;36msection 2: paddr=0x00010d60 vaddr=0x3ffb03f0 size=0x009ac (  2476) load
.[0m.[0;36msection 3: paddr=0x00011714 vaddr=0x40080000 size=0x00400 (  1024) load
.[0m.[0;36msection 4: paddr=0x00011b1c vaddr=0x40080400 size=0x03718 ( 14104) load
.[0m.[0;36msection 5: paddr=0x0001523c vaddr=0x00000000 size=0x0add4 ( 44500) 
.[0m.[0;36msection 6: paddr=0x00020018 vaddr=0x400d0018 size=0x04430 ( 17456) map
.[0m.[0;36mstart: 0x400806e4

.[0mInitializing heap allocator:
Region 19: 3FFB0D9C len 0002F264 tag 0
Region 25: 3FFE8000 len 00018000 tag 1
Pro cpu up.
Pro cpu start user code
ping
ping
ping
ping
ping
```

closed 2016-09-03 by AdySan

igrr, 2016-09-03:
Another possible option — we have changed the "no affinity" constant in
master but wifi libs haven't been updated there yet.

Spritetm, 2016-09-03:
Oooh, that'll break stuff indeed.

AdySan, 2016-09-03:
Yeah, @Spritetm's suggestion kinda fixed the issue too, but I get quite a long trace before getting to the app.

```
ø.ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:2312
load:0x40078000,len:3788
load:0x40098000,len:532
entry 0x4009813c
.[1;36m

.[0m.[1;36m**************************************
.[0m.[1;36m*       hello espressif ESP32!       *
.[0m.[1;36m*        2nd boot is running!        *
.[0m.[1;36m*            version (V0.1)          *
.[0m.[1;36m**************************************
.[0m.[0;36mcompile time 22:35:44

.[0m.[0;36m  SPI Speed      : 40MHz
.[0m.[0;36m  SPI Mode       : QIO
.[0m.[0;36m  SPI Flash Size : 4MB
.[0m.[1;36mPartition Table:
.[0m.[1;36m## Label            Usage          Type ST Offset   Length
.[0m.[1;36m 0 factory          factory app      00 00 00010000 00100000
.[0m.[1;36m 1 rfdata           RF data          01 01 00110000 00040000
.[0m.[1;36m 2 wifidata         WiFi data        01 02 00150000 00040000
.[0m.[1;36mEnd of partition table
.[0m.[1;36mLoading app partition at offset 00010000
.[0m.[0;36msection 0: paddr=0x00000020 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
.[0m.[0;36msection 1: paddr=0x00010010 vaddr=0x3f400010 size=0x085f0 ( 34288) map
.[0m.[0;36msection 2: paddr=0x00018608 vaddr=0x3ffbf790 size=0x0162c (  5676) load
.[0ml;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@`3.@.l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@`3.@.l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@`3.@.l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@`3.@.l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@l;.@`3.@.Initializing heap allocator:
Region 19: 3FFC0DBC len 0001F244 tag 0
Region 25: 3FFE8000 len 00018000 tag 1
Pro cpu up.
Starting app cpu, entry point is 0x40080778
App cpu up.
Pro cpu start user code
erase sector 5
erase sector 6
erase sector 7
nvs_flash_init
misc_nvs_init g_misc_nvs=0x3ffc0f28
frc2_timer_task_hdl:3ffc18c4, prio:22, stack:2048
tcpip_task_hdlxxx : 3ffc33ac, prio:20,stack:2048
Core0 started initializing FreeRTOS. Jumpinwito_icht uwif. 
nit start, config=0x3ffc4880
wifi_mac_init sta 18:fe:34:6a:87:40
wifi_mac_init ap  1a:fe:34:6a:87:40
wifi_nvs_init init nvs start
wifi_nvs_init init nvs - open nvs namespace=nvs.net80211 ok, handle=0x00000002
wifi_nvs_load nvs load:0-0-opmode-0-1-(0-3-2)-0x3ffb02f0-2
wifi_nvs_load, get_u8 fail, key=opmode index=0 not found
wifi_nvs_load use default value=2 ret=0
wifi_nvs_load nvs load:1-1-country-0-1-(0-3-0)-0x3ffb02f1-2
wifi_nvs_load, get_u8 fail, key=country index=1 not found
wifi_nvs_load use default value=0 ret=0
wifi_nvs_load nvs load:2-2-sta.ssid-7-36-(0-0-0)-0x3ffb02f4-2
wifi_nvs_load not found key, default to 0xff
wifi_nvs_load nvs load:3-3-sta.mac-7-6-(0-0-0)-0x3ffb0318-2
wifi_nvs_load not found key, default to 0xff
wifi_nvs_load nvs load:4-4-sta.authmode-0-1-(1-7-1)-0x3ffb031e-2
[... 106 lines cut ...]
ic_set_rx_policy old 600330d8=7155 60033024=1ffff 60033064=1ffff
ic_set_rx_policy new 600330d8=7145 60033024=ffff 60033064=ffff
ic_set_rx_policy bss 0:0
ic_set_rx_policy i=1 beacon=0 bc=0 uc=0
ic_set_rx_policy old 600330dc=7155 6003302c=0 6003306c=0
ic_set_rx_policy new 600330dc=7145 6003302c=0 6003306c=0
ic_set_rx_policy bss 0:0
pp_task_hdl : 3ffc6bfc, prio:23, stack:2048
chip_init s=0
wifi_set_rx_policy set mac i=0 18:fe:34:6a:87:40
wifi_set_rx_policy set mac i=1 1a:fe:34:6a:87:40
wifi_set_rx_policy rx policy i=0 beacon=0 bc=0 uc=0
ic_set_rx_policy i=0 beacon=0 bc=0 uc=0
ic_set_rx_policy old 600330d8=7145 60033024=ffff 60033064=1ffff
ic_set_rx_policy new 600330d8=7145 60033024=ffff 60033064=ffff
ic_set_rx_policy bss 0:0
wifi_set_rx_policy rx policy i=1 beacon=0 bc=0 uc=0
ic_set_rx_policy i=1 beacon=0 bc=0 uc=0
ic_set_rx_policy old 600330dc=7145 6003302c=0 6003306c=1ffff
ic_set_rx_policy new 600330dc=7145 6003302c=0 6003306c=ffff
ic_set_rx_policy bss 0:0
wifi_init init event q=0x3ffc1a3c
wifi_init wifi init done
ping
```

igrr, 2016-09-03:
Some logs to be removed, yeah.

ESP32DE, 2016-09-04:
@AdySan, Guys please :)
the staff have make nice ANSI prints in this - please - use onetime ANSI Term emmulation for the colours in boot and "OCD debugger". :+1:  this would be great! 
best wishes
rudi ;-)

projectgus, 2016-09-06:
master branch should now be updated to avoid the configuration combination that caused this error loop. Thanks for reporting the bug in the first place!