espressif/esp-idf #7: Continuous Rebooting?

hexagon5un, 2016-09-05:
Hi all,

Testing out the ESP32 moduile with the Template App and the IDF, it gets locked into an infinite-reboot cycle.  

Output is like so: 
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3ffc0000,len:0
load:0xffff0e00,len:-1
1162 mmu set 00010000, pos 00010000

With that last block repeating indefinitely.

That 0xffff0e00 address doesn't smell right, but I have no idea where it's coming from.  I have two different modules and both do the same thing.  As far as I can tell, the esptool upload verifies correctly.

I'm stumped.  Any pointers?

Versions:
esp-idf commit is cef73d9 (with the corresponding esp32/libs)
esp-template-app is 3bbf6ce
toolchain is the binary download (on Sep 5)

Spritetm, 2016-09-06:
Hmm, as far as I can tell, that's the ROM trying to load the bootloader. How did you flash the app?

hexagon5un, 2016-09-06:
I flashed with esptool, via `make flash` and the IDF Makefile. 

```
esptool.py v2.0-dev
Erasing flash...
Wrote 7168 bytes at 0x00001000 in 0.7 seconds (82.6 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.91s to erase flash block
Wrote 149504 bytes at 0x00010000 in 14.6 seconds (82.0 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.05s to erase flash block
Wrote 1024 bytes at 0x00004000 in 0.1 seconds (84.0 kbit/s)...
Hash of data verified.
```

I'm using the two small form-factor boards that were sent to Hackaday as demos.  They're labelled ESP_02_20160831-18 and -30.  They worked fine with the SSC commands when they arrived.  I didn't do anything more with the RTOS branch, because IDF looks like the way to go. 

FWIW, the behavior changes when I accidentally touched a few of the pins on the side near the USB connector.  Playing with it intentionally just now, I can change the 2nd load address by touching any of the `SD0..SD3` pins with my finger.  For instance, with my finger on the SD0/SD1 side of the board, it repeats:
...
load:0x3ffc0000,len:0
load:0xeeee0400,len:-286331154
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
...

I don't know about the boot process, but it's starting to look like a pin should have a pullup/down or something? (Unless this is normal behavior.)

If this is a one-off hardware bug, and you'd like to debug this offline, you can reach me directly: elliot.williams@hackaday.com.

Spritetm, 2016-09-06:
Well, the good thing is, I acquired one here as well. (Fyi, the boards are very new, I think they arrived a day or so before we shipped off your samples.) Looks like the flash that's on the chips uses a slightly different command to go into qio read mode. The guys testing the boards have some scripts to modify the bootloader for that, but they haven't managed to integrate it into the master branch yet.

For now, a very quick hack that makes it work: modify esp-idf/components/esptool_py/Makefile.projbuild and for the line that contains 'elf2image', add '--flash_mode dio', like such:

`$(APP_BIN): $(APP_ELF)                                                                                                                                         
    $(Q) $(ESPTOOLPY) elf2image --flash_mode dio -o $@ $< `

hexagon5un, 2016-09-06:
Awesome.  That by itself gets me to the single-core kernel error that the other guy had.  Removing low-level WiFi and enabling both cores in FreeRTOS gets me to ping.

Now all I need to do is figure out something to show off with the chip for an article.  Any suggestions?  What's your favorite oddball feature?

ESP32DE, 2016-09-06:
hi elliot,
you can try to flash an example like this: 

http://esp32.com/viewtopic.php?f=18&t=189

so i know, ( you are a lucky boy :)  am have no board just in time on hand  ) the code is right
and you have an example fo aes256 acceleration and you know about a nice feature that can you named by "AES Accelerator" side 112 

http://espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

;-)
best wishes
rudi ;-)

closed 2016-09-06 by hexagon5un

hexagon5un, 2016-09-09:
Just a note for future generations: the DIO-mode ROM flashing made it into the master branch and everything works out of the box now.

Thanks again, Sprite, for the quick hack-around.

ayziat, 2018-05-08:
Hey all,

I've tried to flash my esp32 but it gets locked into a reboot cycle every 2 minutes.

when rebooting the output looks like so:

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5704
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13568
entry 0x400785e8
W (30) boot: PRO CPU has been reset by WDT.
W (30) boot: WDT reset info: PRO CPU PC=0x4008313e
0x4008313e: spi_flash_op_block_func at /Users/aziat/GroheProject/artifacts/esp-idf/components/spi_flash/cache_utils.c:82 (discriminator 1)

W (31) boot: WDT reset info: APP CPU PC=0x40156a50
0x40156a50: get_socket at /Users/aziat/GroheProject/artifacts/esp-idf/components/lwip/api/sockets.c:1814

I (37) boot: ESP-IDF 2018.2.1-76-g02d2e1b4-dirty 2nd stage bootloader
I (44) boot: compile time 11:35:26
I (48) boot: Enabling RNG early entropy source...
I (53) boot: SPI Speed      : 40MHz
I (57) boot: SPI Mode       : DIO
I (61) boot: SPI Flash Size : 4MB
I (66) boot: Partition Table:
I (69) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (84) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (91) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (99) boot:  3 ota_0            OTA app          00 10 00010000 00120000
I (106) boot:  4 ota_1            OTA app          00 11 00130000 00120000
I (114) boot: End of partition table
I (118) boot: No factory image, trying OTA 0
I (123) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x37f48 (229192) map
I (212) esp_image: segment 1: paddr=0x00047f70 vaddr=0x3ffc0000 size=0x03dbc ( 15804) load
I (219) esp_image: segment 2: paddr=0x0004bd34 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/aziat/GroheProject/artifacts/esp-idf/components/freertos/xtensa_vectors.S:1685

I (219) esp_image: segment 3: paddr=0x0004c13c vaddr=0x40080400 size=0x03ed4 ( 16084) load
I (235) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0xc1028 (790568) map
0x400d0018: _flash_cache_start at ??:?

I (514) esp_image: segment 5: paddr=0x00111048 vaddr=0x400842d4 size=0x0f87c ( 63612) load
0x400842d4: r_assert_err at ??:?

I (540) esp_image: segment 6: paddr=0x001208cc vaddr=0x400c0000 size=0x00000 (     0) load
I (552) boot: Loaded app from partition at offset 0x10000
I (552) boot: Disabling RNG early entropy source...
I (553) cpu_start: Pro cpu up.
I (556) cpu_start: Starting app cpu, entry point is 0x40080f70
0x40080f70: call_start_cpu1 at /Users/aziat/GroheProject/artifacts/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (567) heap_init: Initializing. RAM available for dynamic allocation:
I (574) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (580) heap_init: At 3FFCFC50 len 000103B0 (64 KiB): DRAM
I (586) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (592) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (599) heap_init: At 40093B50 len 0000C4B0 (49 KiB): IRAM
I (605) cpu_start: Pro cpu start user code
I (287) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

any ideas?

thank you all.

Kashyapkoshti, 2018-11-14:
Shows on arduino serial monitor

rst:1x10(RTCWDT_RTC_RESET) , boot:0x1f(SPI_FAST_FLASH_BOOT)
flash read err,1000
ets_main.c 371
ets jun 8 2016 00:22:57

chalengr89, 2026-10-20:
Hello. I can see there’s a problem with load:0xffbf0000, len:-1
since 2016.
Please could you help me?
```
15:04:22.132 -> rst:0x7 (TG0WDT_SYS_RESET), boot:0x13 (SPI_FAST_FLASH_BOOT)
15:04:22.132 -> configsip: 0, SPIWP:0xee
15:04:22.132 -> clk_drv:0x00, q_drv:0x00, d_drv:0x00, cs0_drv:0x00, hd_drv:0x00, wp_drv:0x00
15:04:22.132 -> mode:DIO, clock div:2
15:04:22.132 -> load:0x3fff0030,len:1184
15:04:22.132 -> load:0xffbf0000,len:-1
15:04:22.164 -> 1150 mmu set 00010000, pos 00010000
15:04:22.164 -> 1150 mmu set 00020000, pos 00020000
15:04:22.164 -> 1150 mmu set 00030000, pos 00030000
15:04:22.201 -> 1150 MMU set 00040000, pos 00040000
15:04:22.201 -> 1150 MMU set 00050000, pos 00050000
15:04:22.232 -> 1150 MMU set 00060000, pos 00060000
15:04:22.232 -> 1150 MMU set 00070000, pos 00070000
15:04:22.232 -> 1150 MMU set 00080000, pos 00080000
15:04:22.265 -> 1150 MMU set 00090000, pos 00090000
15:04:22.265 -> 1150 MMU set 000a0000, pos 000a0000
15:04:22.300 -> 1150 MMU set 000b0000, pos 000b0000
15:04:22.300 -> 1150 MMU set 000c0000, pos 000c0000
15:04:22.333 -> 1150 MMU set 000d0000, pos 000d0000
15:04:22.333 -> 1150 MMU set 000e0000, pos 000e0000
15:04:22.333 -> 1150 MMU set 000f0000, pos 000f0000
15:04:22.369 -> 1150 MMU set 00100000, pos 00100000
15:04:22.369 -> 1150 MMU set 00110000, pos 00110000
15:04:22.369 -> 1150 MMU set 00120000, pos 00120000
15:04:22.411 -> 1150 MMU set 00130000, pos 00130000
15:04:22.411 -> 1150 MMU set 00140000, pos 00140000
15:04:22.411 -> 1150 MMU set 00150000, pos 00150000
15:04:22.444 -> ets Jul 29 2019 12:21:46
```

My sketch won’t work.
```

void setup() {
  Serial.begin(115200);
 
  delay(1000); 
  
  Serial.println("--- START ---");
  Serial.println("I am working!");
}

void loop() {
 
}
```

```
esptool.py v4.5.1
Serial port COM3
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:e9:f4:a3:85:80
Uploading stub...
Running stub...
Stub running...
MAC: d4:e9:f4:a3:85:80
Hard resetting via RTS pin...
```