espressif/esp-idf #9: esp-idf build error

EFudd, 2016-09-11:
Performed the build serval times using crosstool-NG and several other toolchains (on latest Debian).  Always receive error on xtensa_context.S line 458 ->  224 is not a valid register for rsr.  
CPENABLE is def'ed as 224.

igrr, 2016-09-11:
I suspect there may be a problem with the toolchain.

Have you followed these instructions to build the toolchain?
https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst#alternative-step-1-compile-the-toolchain-from-source-using-crosstool-ng

What are the "several other toolchains" are you referring to?

If you have built the toolchain using crosstool-NG, please run the following diagnostic command and copy its output here:
```
echo -ne "#include \"xtensa/config/core-isa.h\"\n#pragma message XCHAL_CORE_ID" | xtensa-esp32-elf-gcc -c -o /tmp/1.o -xc -
```

(make sure xtensa-esp32-elf-gcc is in your PATH before running this).

EFudd, 2016-09-13:
Thank you for that rapid respons:

toolchains used in this order:
1) built: github.com/jcmvbkbc/crosstool-NG - esp32 builed failed at 
xtensa_context.S
2) prebuilt toolchain: github.com/jmattsson/nodemcu-prebuilt-toolchains 
- esp32 builed failed at xtensa_context.S
3) built:  github.com/espressif/crosstool-NG - esp32 build failed at 
xtensa_context.S

I tried to isolate the various builds but there must have been some leak 
between them.

Rebuilt github.com/espressif/crosstool-NG on a fresh install of Debian 
Jessie.
     The esp32-idf builds were succesful.

Unfortunately, mix & match with tool chains doesn't work very well.

Regards, E.F.

igrr, 2016-09-13:
Great to know that building from https://github.com/espressif/crosstool-NG worked for you!

Just a note to other people who may be reading this: Espressif provides prebuilt toolchains for all platforms, links may be found in [docs/linux-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst), [docs/macos-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/macos-setup.rst), [docs/windows-setup.rst](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst).

Toolchain in https://github.com/jmattsson/nodemcu-prebuilt-toolchains is not guaranteed to work.

closed 2016-09-13 by igrr