espressif/esp-idf #10: project build error

hardikharpal, 2016-09-13:
[image]

igrr, 2016-09-14:
Can you please do `cat /c/ESP/esp-idf/tools/kconfig/zconf.gperf` and post the output?

hardikharpal, 2016-09-14:
here it is,
`Hardik Harpal@Hardik MSYS /c/esp-idf-template
$ cd C:/esp-idf-template

Hardik Harpal@Hardik MSYS /c/esp-idf-template
$ cat /c/ESP/esp-idf/tools/kconfig/zconf.gperf
%language=ANSI-C
%define hash-function-name kconf_id_hash
%define lookup-function-name kconf_id_lookup
%define string-pool-name kconf_id_strings
%compare-strncmp
%enum
%pic
%struct-type

struct kconf_id;

static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);

%%
mainmenu,       T_MAINMENU,     TF_COMMAND
menu,           T_MENU,         TF_COMMAND
endmenu,        T_ENDMENU,      TF_COMMAND
source,         T_SOURCE,       TF_COMMAND
choice,         T_CHOICE,       TF_COMMAND
endchoice,      T_ENDCHOICE,    TF_COMMAND
comment,        T_COMMENT,      TF_COMMAND
config,         T_CONFIG,       TF_COMMAND
menuconfig,     T_MENUCONFIG,   TF_COMMAND
help,           T_HELP,         TF_COMMAND
---help---,     T_HELP,         TF_COMMAND
if,             T_IF,           TF_COMMAND|TF_PARAM
endif,          T_ENDIF,        TF_COMMAND
depends,        T_DEPENDS,      TF_COMMAND
optional,       T_OPTIONAL,     TF_COMMAND
default,        T_DEFAULT,      TF_COMMAND, S_UNKNOWN
prompt,         T_PROMPT,       TF_COMMAND
tristate,       T_TYPE,         TF_COMMAND, S_TRISTATE
def_tristate,   T_DEFAULT,      TF_COMMAND, S_TRISTATE
bool,           T_TYPE,         TF_COMMAND, S_BOOLEAN
boolean,        T_TYPE,         TF_COMMAND, S_BOOLEAN
def_bool,       T_DEFAULT,      TF_COMMAND, S_BOOLEAN
int,            T_TYPE,         TF_COMMAND, S_INT
hex,            T_TYPE,         TF_COMMAND, S_HEX
string,         T_TYPE,         TF_COMMAND, S_STRING
select,         T_SELECT,       TF_COMMAND
range,          T_RANGE,        TF_COMMAND
visible,        T_VISIBLE,      TF_COMMAND
option,         T_OPTION,       TF_COMMAND
on,             T_ON,           TF_PARAM
modules,        T_OPT_MODULES,  TF_OPTION
defconfig_list, T_OPT_DEFCONFIG_LIST,TF_OPTION
env,            T_OPT_ENV,      TF_OPTION
allnoconfig_y,  T_OPT_ALLNOCONFIG_Y,TF_OPTION
%%

Hardik Harpal@Hardik MSYS /c/esp-idf-template
$ make V=1
MAKEFLAGS="" \
CFLAGS="" \
LDFLAGS="" \
CC=cc LD=ld \
make -C C:/ESP/esp-idf/tools/kconfig
make[1]: Entering directory '/c/ESP/esp-idf/tools/kconfig'
gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t zconf.gperf
zconf.gperf:14: warning: junk after %% is ignored
, defaulting to Cage option ANSI-C
zconf.gperf:5: junk after declaration
make[1]: *** [Makefile:300: zconf.hash.c] Error 1
make[1]: Leaving directory '/c/ESP/esp-idf/tools/kconfig'
PROJECT_PATH= \
LDFLAGS= \
CFLAGS= \
BUILD_DIR_BASE=/c/esp-idf-template/build/bootloader \
make -C C:/ESP/esp-idf/components/bootloader/src MAKEFLAGS= V=1 TARGET_BIN_LAYOUT="" /c/esp-idf-template/build/bootloader/bootloader.bin
make[1]: Entering directory '/c/ESP/esp-idf/components/bootloader/src'
MAKEFLAGS="" \
CFLAGS="" \
LDFLAGS="" \
CC=gcc LD=ld \
make -C C:/ESP/esp-idf/tools/kconfig
make[2]: Entering directory '/c/ESP/esp-idf/tools/kconfig'
gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t zconf.gperf
zconf.gperf:14: warning: junk after %% is ignored
, defaulting to Cage option ANSI-C
zconf.gperf:5: junk after declaration
make[2]: *** [Makefile:300: zconf.hash.c] Error 1
make[2]: Leaving directory '/c/ESP/esp-idf/tools/kconfig'
make[2]: Entering directory '/c/esp-idf-template/build/bootloader/main'
gcc -DESP_PLATFORM -Og -std=gnu99 -g3 -Wpointer-arith -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -Wall -ffunction-sections -fdata-sections  -MMD -MP -I /c/ESP/esp-idf/components/bootloader/src/main/include -I /c/ESP/esp-idf/components/bootloader/src/build/include/ -I C:/ESP/esp-idf/components/esp32/include -I. -c /c/ESP/esp-idf/components/bootloader/src/main/./flash_encrypt.c -o flash_encrypt.o
gcc: error: unrecognized command line option ‘-mlongcalls’
make[2]: *** [C:/ESP/esp-idf/make/component_common.mk:113: flash_encrypt.o] Error 1
make[2]: Leaving directory '/c/esp-idf-template/build/bootloader/main'
make[1]: *** [C:/ESP/esp-idf/make/project.mk:214: main-build] Error 2
make[1]: Leaving directory '/c/ESP/esp-idf/components/bootloader/src'
make: *** [C:/ESP/esp-idf/components/bootloader/Makefile.projbuild:20: /c/esp-idf-template/build/bootloader/bootloader.bin] Error 2
`

hardikharpal, 2016-09-14:
[image]
[image]

projectgus, 2016-09-14:
Hi @hardikharpal ,

This looks very similar to another problem we're investigating on Windows. There's something unexpected going on with some paths in MSYS.

If you move the esp-idf & template directories to /home/yourusername inside msys (in Windows this is C:\msys32\home\yourusername), then it should work. I'll let you know once we have a proper fix.

hardikharpal, 2016-09-14:
Thank you @projectgus   @igrr 
I can do that as my user name has space in between that's why I guess its showing error.
I will try with other username(without space) and let you know.
Hardik Harpal@Hardik MSYS ~
$  cd C:/msys32/home/Hardik Harpal/esp-idf
bash: cd: C:/msys32/home/Hardik: No such file or directory

hardikharpal, 2016-09-14:
Thank you @igrr @projectgus 
@projectgus  I tried as you said its showing following error.As my user name contain space I had to add backslash to escape each space within path while using msys32. 
cd C:/msys32/home/**Hardik\ Harpal**/esp-idf-template instead of 
cd C:/msys32/home/**Hardik Harpal**/esp-idf-template

due to space in path its not working in some setups. Is it?
 
[image]

hardikharpal, 2016-09-14:
with eclipse 
[image]

projectgus, 2016-09-16:
Thanks for working through all this and providing such comprehensive screenshots.

* The original bug is caused if git clones the local repository with CRLF (Windows-style) line endings. The git we provide (part of MSYS2) doesn't use these by default, but many other git for Windows setups do.

There is a fix coming soon so the build system will work with CRLF line endings, but in the meantime if you re-clone with LF (Unix-style) endings then things should work. You may need to explicitly run `/usr/bin/git clone` in order to get the right version of git.

* You are correct that esp-idf doesn't like spaces in paths. There isn't a lot we can do about this ("make" makes a lot of assumptions about spaces). Sorry for misleading you with the previous suggestion. I've added a note to the docs (also in the pipeline) that project & esp-idf paths should not contain spaces.

Thanks again for your help with this. The issue will be updated when the fixes reach github.

hardikharpal, 2016-09-16:
Thank you for detailed explanation and temporary solution to my issue.

igrr, 2016-09-27:
Support for CRLFs is now present in master.

closed 2016-09-27 by igrr

clachance, 2018-04-06:
Hi,  

Is there a way to prevent mingw32 to use the current windows user to create the directory in Home?  The thing is that if your windows username have space (in my case : "Christian Lachance"), it will create problems later.   How can i change the folder to something without space?  I tried to rename the folder to Christian_Lachance but it just make it worst.

Thanks!

Christian.

olivierzign, 2018-06-15:
To trick mingw32.exe used the environment variable HOME to create the home dir. So change this to the new home dir location without spaces and rerun mingw32.exe.