espressif/esp-idf #11: Installation step 2: not a valid identifierline 89: export: `sha1

krzychb, 2016-09-15:
Hi Team,

Following [Step 2: Getting the esp-idf repository from github](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst#step-2-getting-the-esp-idf-repository-from-github) of the installation process on Windows 7 x64,  I got the following two messages regarding sha1 (at the end of the full log below):

```
$ git clone --recursive https://github.com/espressif/esp-idf.git
Cloning into 'esp-idf'...
remote: Counting objects: 3162, done.
remote: Total 3162 (delta 0), reused 0 (delta 0), pack-reused 3162
Receiving objects: 100% (3162/3162), 5.05 MiB | 1.03 MiB/s, done.
Resolving deltas: 100% (1695/1695), done.
Checking connectivity... done.
Submodule 'components/esp32/lib' (https://github.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp32/lib'
Submodule 'components/esptool_py/esptool' (https://github.com/themadinventor/esptool.git) registered for path 'components/esptool_py/esptool'
Cloning into '/esp-idf/components/esp32/lib'...
Cloning into '/esp-idf/components/esptool_py/esptool'...
': not a valid identifierline 89: export: `sha1
': not a valid identifierline 89: export: `displaypath
Submodule path '': checked out '1303c92c1056b7f59b95360b58e70a21cb4a93e1'
': not a valid identifierline 89: export: `sha1
': not a valid identifierline 89: export: `displaypath
Submodule path '': checked out '7c84dd433512bac80e4c01c569e42b4fe76646a7'
```
I have tried this step twice with the same result.

The `/esp-idf/components/esptool_py/esptool` directory contains some files and sub directories.
I did not find any leads to this error on `https://github.com/themadinventor/esptool` repository.

Should I be concerned?

hardikharpal, 2016-09-15:
Hi @krzychb,
Until u got solution from esp32 team,try cloning git by desktop git app.I had some issue to but with desktop git app it worked.

hardikharpal, 2016-09-15:
Hi @krzychb,
Until u got solution from esp32 team,try cloning git by desktop git app.I had some issue too but with desktop git app it worked.

igrr, 2016-09-15:
Seen this message on colleague's windows computer as well. It looked like everything was actually checked out correctly.
Here's a similar error mentioned in msys2 tracker: https://github.com/Alexpux/MSYS2-packages/issues/30. My guess that that's something with msys2 git, we'll check it...

projectgus, 2016-09-15:
Hi @krzychb,

Quick question, what directory were you in when you ran `git clone`? (ie what was the output of the `pwd` command?) Anything special about that drive (network mapped drive, removable drive, etc.)?

Also, what is the output of the command `which git`?

Thanks,

Angus

projectgus, 2016-09-16:
(Deleted previous comment because I can reproduce this without any special directories, etc.)

I can confirm @igrr's comment that despite this error everything seems to work fine and the git submodule checkout is 100% valid. So it's nothing you have to worry about for now, provided esp-idf works as expected.

I did notice this path is directly in the msys "root":
```
Cloning into '/esp-idf/components/esp32/lib'...
```

Did you mean to run the git checkout from the directory `/`? (this corresponds to c:\msys32 in Windows). It's OK if you did, just wanted to be sure this isn't unexpected behaviour.

krzychb, 2016-09-16:
@hardikharpal, @igrr, @projectgus - thank you guys for really quick replies and support.

@projectgus - indeed I did git checkout intentionally from '/'.

From deciding to set up the `esp-idf` to getting `esp-idf-template` compiled, took me maybe 45 minutes.  To get to this point with ESP8266 1.5 years ago has not been so easy for me and I was not sure what to expect now from ESP32 framework.  ESP32 Team made it it really accessible and easy to follow.

Congratulations :smile: :+1:

closed 2016-09-16 by krzychb

projectgus, 2016-09-16:
Thanks for the kind words, that's exactly the kind of result we are all aiming for!

I've added a note in our Windows documentation about the weird submodule messages, with a link to this issue. There may eventually be a solution, but at least it's not breaking any functionality.