espressif/esp-idf #14: Flash problem

feelfreelinux, 2016-09-17:
Hi, i'm trying to flash 'myapp' template to esp32. But when i'm using dio mode, i get 

> A fatal error occurred: Failed to enter RAM download mode (result was 0xff, 0x0)
But, when i'm using the qio mode it flash sucessfully, but i'm getting resets from issue 7 (https://github.com/espressif/esp-idf/issues/7)

projectgus, 2016-09-19:
Hi @feelfreelinux,

Thanks for reporting this. Are you using a ESP32 "Core" board with onboard USB?

Can you post the full esptool.py command line and output from esptool.py, please? If you're using a `make flash` type target, you can run `make flash V=1` to see the full command line.

projectgus, 2016-09-21:
We've reproduced this bug internally and there's a fix on its way.

In the meantime, you can ignore the error - the data is flashed 100% successfully in both cases.

igrr, 2016-09-27:
This should be now fixed in master. Please do a `git pull origin master` followed by `git submodule update` to get updated esptool.py.

closed 2016-09-27 by igrr