espressif/esp-idf #17: Unresolved inclusion: "freertos/FreeRTOS.h"

krzychb, 2016-09-19:
Hey Team,

The [Toolchain for Windows: Quick Steps](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst#step-1-toolchain-for-windows-quick-steps) works for me and I am able to compile the [Getting Started project](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst#step-3-starting-a-project) without any issues or tweaking standard configuration.

Then I have installed from scratch and configured Eclipse following instructions in [Setting up Eclipse](https://github.com/espressif/esp-idf/blob/master/docs/eclipse-setup.rst#setting-up-eclipse).

This one did not work for me out of box:
[image]

To fix this issue I have entered:
[image]

Did I miss something in installation process or this is intended / obvious user should add this missing include path? Or maybe it should be added somewhere else?

projectgus, 2016-09-19:
Thanks for reporting this with so much detail.

There is an issue on Windows with Eclipse CDT parsing output from the build, that causes these errors. We're working on a solution.

In the meantime, you should find that Project -> Build works despite the errors (the errors are caused by the IDE parsing output from the build for its internal information about where the header files are, not by the building itself.)

Will update the ticket when we have a fix.

krzychb, 2016-09-20:
@projectgus, thank you for quick review and providing the issue background. Project build is indeed successful (seeing the errors I did not think about checking the "Console" tab). I was not aware that Eclipse has such "split personality". 

I like Eclipse for being able to flag issues before I even think about building a project. On the other hand I do not like false, distracting alarms. Therefore, before the fix is identified, I will keep additional path in the "Include directories" so bogus errors are not reported.

mseerig, 2016-11-30:
I have also problems with includes and symbols. Following the instructions in Setting up Eclipse gave me exactly the result which @**krzychb** also had. 

`#include "freertos/FreeRTOS.h"
#include "freertos/task.h"`

... is now included manually by the project properties. 

[image]

For me, as a beginner:
What have I to do, that a compilation the code will success in Eclipse? 
How can I update the Index in Eclipse?
-> I want to open the function `esp_restart();` with pressing **F3**..

krzychb, 2017-01-17:
I am afraid, I do not see this issue fixed with a14bef8f8a854be557d40c893af513f963e1cac4 and the latest toolchain.

Within the same installation, if I checkout branch `bugfix/eclipse_windows_paths`, and try again, the issue is gone. If I go back to `master`, it is back.

Bendito999, 2017-01-17:
I'm not sure if this is your exact problem, but: I just setup eclipse, the extra thing I had to do that was not included in the official setup instructions was under C/C++ general, under the "Paths and Symbols", I had to add:
${IDF_PATH}/components/esp32/include
${IDF_PATH}/components/newlib/include
${IDF_PATH}/components/freertos/include
${IDF_PATH}/components/nvs_flash/include
${IDF_PATH}/components/driver/include
${IDF_PATH}/components/log/include
Set the last one with a "workspace" checkbox, replace esp-idf-template with your project folder name
/esp-idf-template/build/include
Hope that this helps!

krzychb, 2017-01-18:
Thanks, @Bendito999. Yes, this is the problem and it affects Windows users. The good news is that there are vistas to get it fixed. Solution has been prepared by @projectgus in `bugfix/eclipse_windows_paths`. In the meantime there is another workaround provided  by @motla in https://github.com/espressif/esp-idf/pull/157#issuecomment-267269741

grisevg, 2017-01-26:
@krzychb but I have the same issue on Linux. Includes are not recognised by Eclipse unless I add them manually.

krzychb, 2017-02-22:
Resolved in https://github.com/espressif/esp-idf/commit/8dede8f8a46f42661665e9b924b7f455d06ef00a

dziekon, 2017-09-15:
The issue is not resolved for C++ files, as suggested pattern:

xtensa-esp32-elf-(g?cc)|([gc]\+\+)|(clang)

doesn't match "xtensa-esp32-elf-c++". The existing matcher should be wrapped with additional group:

xtensa-esp32-elf-((g?cc)|([gc]\+\+)|(clang))

See: https://github.com/espressif/esp-idf/pull/1005

projectgus, 2017-09-20:
@dziekon We can probably treat #1005 as a separate issue (as it's a C++-specific not Windows-specific problem).  The fix in that PR should be merged soon.

closed 2017-09-20 by projectgus

fralomb, 2017-12-09:
I still have the same problem on Eclipse Oxigen for Mac OS X (using C++). I've tried to modify the compiler command pattern as suggested from @dziekon and to add the the several paths suggested by @Bendito999 but the "freertos/FreeRTOS.h" and "freertos/task.h" are still unresolved in my project..

krzychb, 2017-12-09:
@fralomb. I suggest to open a separate issue and indicate in the title that the issue concerns  C++, Mac OS X and Eclipse Oxigen. This should focus the attention of  C++ / Mac OS X users that may face similar issue. Also, have you tried to do *Index > Rebuild* from the context menu of affected the project?

EricAtPrendo, 2018-02-20:
I think this issue is still here, I am running Ubuntu on a virtual machine.  I followed the current instructions for building an Eclipse environment, but have Unresolved inclusion errors even though it compiles OK.

[image]

ronald245, 2018-03-27:
I can confirm the error is still there, 
after following the setup guide word for word, eclipse is still marking the inclusions as unresolved whilst the build itself works just fine.

[image]

nero01, 2018-04-12:
Hello i can confirm the same issue as @ronald245, i used the exact same procedure as described on the official ESP-IDF eclipse guide and also followed the instruction to add the includes .
I am running the latest eclipse on OSX and still have the issue that all includes are shown as unresolved. I can still build and flash the project, even if all includes show as unresolved.

chegewara, 2018-04-12:
I cant tell where the issue is but i have eclipse on Ubuntu 14.10 (vbox) and 3 or 4 different workspaces, from which only one does not have issue with indexing included header files. I have no freaking idea what did i do to make it work, because every next workspaces cant find and index header files even if there is no issue with compiling app.

ahmedawais, 2018-06-03:
running on windows 10
eclipse oxygen 
C:/esp32/pcb/espressif-master/esp32/app/ESP32_ble_UART/main/bta_api.h:28:23: fatal error: bt_target.h: No such file or directory
compilation terminated.

please help i have included all files and bta_api is present in path

sabbiolino80, 2019-04-28:
It seems to me that after 3 years the issue is still there.
Eclipse 4.11 on Ubuntu 18.04.2 VM. I followed the get-started guide. Includes are resolved in the main.c file but not in other files.

sabbiolino80, 2019-04-28:
[image]
this is the quick fix

bonmotwang, 2019-04-28:
Same issue is still there.
[image]

projectgus, 2019-04-29:
Hi @sabbiolino80 @bonmotwang ,

Please check carefully that all of the Project Properties are set exactly as described here:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/eclipse-setup.html#project-properties

If any of the environment variables, "Providers", or "Indexer" settings are wrong then the build may succeed but Eclipse won't be able to parse the build output to learn the include paths for each source file.

If changing any of these options, you may need to do a clean build and then possibly [rebuild the Eclipse index](https://github.com/espressif/esp-idf/issues/17#issuecomment-381159674) in order to see any update in the editor window.

sabbiolino80, 2019-04-29:
@projectgus I did all the steps you mentioned before posting

projectgus, 2019-04-29:
@sabbiolino80 Could you zip up the `.cproject` and `.project` files, and the `.settings` directory from your IDF project and attach the zip file to the issue? There shouldn't be anything more sensitive than path names in these files.

akotowski, 2019-04-30:
I have this problem also.  Sometimes adding an explicit library paths helps, sometimes not.
From the console I notice that when it can't find a file the path to the workspace has been changed to all lower case letters as below:

In file included from c:\users\andreas\eclipse-workspace_old\wm42e\components\async-mqtt-client\src\asyncmqttclient.h:4:0,
                 from C:/Users/Andreas/eclipse-workspace_old/WM42E/main/WM42E.cpp:17:
c:\users\andreas\eclipse-workspace_old\wm42e\components\async-mqtt-client\src\AsyncMqttClient.hpp:21:37: fatal error: AsincMqttClient/Flags.hpp: No such file or directory
compilation terminated.
make[1]: *** [/home/Andreas/esp/esp-idf/make/component_wrapper.mk:286: WM42E.o] Error 1
make[1]: Leaving directory '/c/Users/Andreas/eclipse-workspace_old/WM42E/build/main'
make: *** [C:\msys32\home\Andreas\esp\esp-idf/make/project.mk:530: component-main-build] Error 2
"make all" terminated with exit code 2. Build might be incomplete.

kmmaran, 2019-05-15:
@projectgus hi mate, not sure what is the issue. I am able to compile and run example files, but I can't make any changes to the file. For example
printf("Hello hello hello\n");
but output always
Hello World! (this is the initial Printf)

I have no issue with eclipse setup or sort of things. 
I have tried to build Interrupt GPIO from examples, (from example) GPIO 18 and 19, interrupt to GPIO 4 and 5, but if I change it to say GPIO 16 and 17 to GPIO 4 and 5, it never works, everything build and flashed but, it still running Interrupt from 18 and 19. Why, I cannot understand this , no matter which GPIO I change for IN and OUT, it always go back to GPIO 18 and 19, interrupt to GPIO 4 and 5.

How do I solve this, the incremental build is success but not working on the board.

projectgus, 2019-05-16:
Hi @kmmaran ,

Please ask these kind of unrelated questions on new forum threads (esp32.com) or by opening a new issue for each one.

For the first question, are you sure you are building and the flashing the same project your are editing?

For the second question, please post on the esp32.com forum and include your modified code.

(Please don't reply to this issue about these two things, if you still have questions then follow up in one of the other places I mentioned.)

dsptech, 2019-05-16:
Hi @sabbiolino80,
I experienced the same issue time ago. The only way that fully worked for me was adding a specific make target for the compiler settings provider.

1) My additional makefiles: 
https://secure.dsplabs.net/Cloud/s/SqpSHJP7F853HRX

2) Append in your project makefile: 
"include /c/MyPrj/batch/espidfSettingsTarget.mk"
(change the path to your proper location)

3) In Eclipse Project settings, enable the "CDT GCC Built-in Compiler Settings" and set the command:
"python ${IDF_PATH}/tools/windows/eclipse_make.py getsettings  --silent 'V=' 'SPECSFILE=${INPUTS} '" 

 4) Disable any output parser from the project provider list

I hope that it could be useful to you (I never tested it with a new project by the latest ESP-IDF framework)
Regards

akotowski, 2019-05-16:
Hi @dsptech:
Your fix works!
Thank you very much!
Andy Kotowski

dsptech, 2019-05-16:
Hi @akotowski ,
I'm glad to hear that. Thank you for the feedback.
Regards.

sdrshnptl, 2019-07-07:
> I'm not sure if this is your exact problem, but: I just setup eclipse, the extra thing I had to do that was not included in the official setup instructions was under C/C++ general, under the "Paths [...]

That's the excellent solution!!!
Thanks mate!

OmkarPrabhu08, 2019-07-12:
I am working on HFP code which includes some libraries from Bluedroid folder in components section. Now I have changed component specific configs from make menuconfig as suggested by few of the discussions on ESP ADF/IDF github issues but still the problem hasn't solved completely. Although linker started linking few of the Bluedroid components it's not able to link all. So what configs should I use in that case to run HFP? 
[image]
[hfp_logs.txt](https://github.com/espressif/esp-idf/files/3387480/hfp_logs.txt)
[image]
[image]
[image]

I am attaching my logs for reference and directories path, menuconfig screenshots and profile.d. I am quite sure that the paths are set right cause I am able to compile and run demo codes. Let me know if you need any more info about the issue. Thanks once again.

RomanLutig, 2019-10-11:
Hi to all 
I was able to fix this by changing the path style in the export_idf_path.sh and export_adf_path.sh from "C:/msys32/home/user-name/esp/esp-idf" to "/c/msys32/home/user-name/esp/esp-idf".
[image]

Also, I made the same in the project setting, I changed the style in IDF_PATH and ADF_PATH from "C:\msys32\home\user-name\esp\esp-idf" to "c:\msys32\home\user-name\esp\esp-idf".
Use the lowercase "**c**" character instead of uppercase "**C**"
[image]

I was checking this for all my projects which had this issue before:
1. Change the path style
2. Clean project
3. Build project

After those 3 steps, all "include" error disappears.
[image]

Maybe it will help someone.

Sterpa, 2019-11-15:
Try to do all in this topic.
In the end got last errors:
```
Field 'cores' could not be resolved
Field 'features' could not be resolved
Function 'BIT' could not be resolved
```
Oh, could anybody help with idea?
[image]

RomanLutig, 2019-11-27:
Check this mark in the project settings.
[image]

davidlehrian, 2021-08-21:
Hello,
Searches for my issue keep pointing me back to this old closed issue.  I am using esp-idf from the master branch with Eclipse.  I am able to create and build projects and have successfully flashed devices so the environment appears to be set up correctly.  I am having a problem whereby my project is compiling correctly however Eclipse is showing me an error marker.  I have pared the code down to a simple reproducible example.  I started by creating a project using the sample_project template.  First I created a function that uses a few esp-idf methods.

Here is test_function.h
```
/* this is a test file */
#include <esp_system.h>
void test_function_a();
```

Here is test_function.c
```
#include <freertos/FreeRTOS.h>
#include <driver/gpio.h>
#include "test_function.h"

void test_function_a(){
	gpio_pad_select_gpio(26);
	gpio_set_direction(26, GPIO_MODE_OUTPUT);
	gpio_set_level(26,0);
}
```

This compiled and runs fine and Eclipse seems to find the function calls and does not give me any error markers.  I wanted to turn these functions into a cpp class so I created the following class header TestClass.h.

```
/* test class */
#include <esp_system.h>

#ifndef CLASS_TestClass_H_
#define CLASS_TestClass_H_

class TestClass {
private:
	TestClass();
	uint8_t GPIO;
public:
	TestClass(uint8_t pin);
	void testMethod();
};
#endif
```

And here is TestClass.cpp
```
/* test class */
#include <freertos/FreeRTOS.h>
#include <driver/gpio.h>
#include "TestClass.h"

TestClass::TestClass(uint8_t pin){
	GPIO=pin;
}

void TestClass::testMethod(){
	gpio_pad_select_gpio(GPIO);
	gpio_set_direction((gpio_num_t)GPIO, GPIO_MODE_OUTPUT);
	gpio_set_level((gpio_num_t)GPIO,0);
}
```

The issue is that, while everything compiles correctly, Eclipse is giving me error markers for the three functions calls in testMethod.

[image]

And simply commenting out #include <freertos/FreeRTOS.h> makes the error markers go away.

[image]

I know I'm not using anything out of <freertos/FreeRTOS.h> in this example, but I was trying to use taskENTER_CRITICAL and taskEXIT_CRITICAL which do require its inclusion.  

The three error markers give me the following errors when I hover over them:
 1st marker:

> Problem description: Invalid arguments ' Candidates are: void gpio_pad_select_gpio(unsigned char) '

2nd marker: 
1st error:

> Problem description: Invalid arguments ' Candidates are: int gpio_set_direction (enum {C:\Users\lehrian\Documents\esp-idf\components\hal\include\hal\gpio_types.h:7814}, enum {C: \Users\lehrian\Documents\esp-idf\components\hal\include\hal\gpio_types.h: 20385}) '

2nd error:

> Type 'gpio_num_t' could not be resolved

3rd error:

> Symbol GPIO_MODE_OUTPUT'' could not be resolved

3rd marker:
1st error:

> Problem description: Invalid arguments ' Candidates are: int gpio_set_level (enum {C:\Users\lehrian\Documents\esp-idf\components\hal\include\hal\gpio_types.h:7814}, unsigned int) '

2nd error:

>  Type 'gpio_num_t' could not be resolved