espressif/esp-idf #18: 25 warnings for the Getting Started project

krzychb, 2016-09-20:
Running `make` on [Toolchain for Windows](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst#step-1-toolchain-for-windows-quick-steps) for the bare [Getting Started project](https://github.com/espressif/esp-idf/blob/master/docs/windows-setup.rst#step-3-starting-a-project) completes successfully and shows 25 warnings. Below is the list taken from Eclipse for easier reading:

[image]

Do you see the same?

igrr, 2016-09-20:
Yes, we also see this. Will clean up.

igrr, 2016-09-26:
Fixed in master, compiling the template app shouldn't produce any warnings now.

closed 2016-09-26 by igrr

dsptech, 2018-08-14:
I mistyped the issue number, so, I erroneusly regferenced this issue. Sorry.
dsptech