espressif/esp-idf #19: OSX build issue gperf unrecognized % directive

annejan, 2016-09-21:
I'm trying to start a project based on the **esp-idf-template**

Unfortunately the configuration step fails.
```
✘-2 ~/Projects/shabadge/esp-eink [master|✚ 1] 
10:57 $ make menuconfig
MAKEFLAGS="" \
	CC=cc LD=ld \
	/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/annejan/Projects/shabadge/esp-idf/tools/kconfig
sed -E "s/\r//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
(standard input):4: unrecognized % directive
make[1]: *** [zconf.hash.c] Error 1
make: *** [/Users/annejan/Projects/shabadge/esp-idf/tools/kconfig/mconf] Error 2
```
I'm using GNU gperf 3.0.4
Let me know if you require any more information .

igrr, 2016-09-21:
Thanks for the report.
We have a fix for this issue, it should reach github soon. Until this happens, you can make the change locally:

```diff
-	sed -E "s/\r//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$$' -p -t
+	sed -E "s/\\x0D$$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$$' -p -t
```

annejan, 2016-09-21:
Thank you, that got me a step further . .

I also had to add `-lncurses` to the `LOADLIBES_mconf`

Now the menuconfig works like a charm . .

igrr, 2016-09-21:
Okay, i'll check the bit about `-lncurses`. Do you use homebrew or macports to install packages? In case of homebrew, do you also have pkgconfig installed?

annejan, 2016-09-21:
Using homebrew on macOS Sierra (10.12 Beta (16A313a))

annejan, 2016-09-21:
Fixed in af82eed35d775613b0d3b6bb0fe50f7417c700f8

closed 2016-09-21 by annejan