espressif/esp-idf #22: esp-idf/components/bt/.//bt.c', needed by 'bt.o'. Stop.

ESP32DE, 2016-09-24:
*** No rule to make target 'c:/sdk32/esp-idf/components/bt/.//bt.c', needed by 'bt.o'. Stop.
hi guys
get this if i compile a simple hello world.
after i rename in esp-idf/components/bt/bt.c  to esp-idf/components/bt/bt.c_  
[image]
compiles without stop.

not sure what this can be ( path err or include err ) 
have seen just in time.

best wishes
rudi ;-)

igrr, 2016-09-24:
Have you tried `make clean` and then `make` again?

ESP32DE, 2016-09-24:
hi ivan
txs for reply
yes - i did

ESP32DE, 2016-09-24:
dir /s /B c:\sdk32\esp-idf > sdk32.tree.txt

[sdk32.tree.txt](https://github.com/espressif/esp-idf/files/491058/sdk32.tree.txt)

ESP32DE, 2016-09-24:
C:\college\esp-devF>echo %IDF_PATH%
c:/sdk32/esp-idf

ESP32DE, 2016-09-24:
just in time tested the 04 example bt app
the same is if the bt.c is in the components folder
'c:/sdk32/esp-idf/components/bt/'
if the file bt.c renamed to bt.c_ then there is no stop but find then the extern define not

extern void btdm_osi_funcs_register(void *osi_funcs);
extern void btdm_controller_init(void);

then i tested this:
i copied the bt.c in the example bt app that can find the extern 
and now compiled without stop and without mistake
and yes, the example bt flashed and run.. i must search my BLE handy now  ;-))
( you guys did it ;-)   ) 
 
[image]

i tested this same procedure in the hello world without ble support.
i copied the bt.c from 
'c:/sdk32/esp-idf/components/bt/'
to the main hello world folder

no compile without stop and without mistake.

think there is a path error or missing path define for component bt.c

btw: component driver/gpio.c
the include for gpio.h is named : driver/gpio.h and 
the include folder has a subfolder "driver" with the gpio.h file.

the component bt/bt.c 
the include for bt.h is named only : bt.h
and the include folder does not have a subfolder:  bt/bt.h
the bt.h is in the "include" folder

not sure is this ok.

the second i saw is:
the lib folder is in the components bt folder
is this right here? 
does we not move the lib to the system lib folder?

txs ivan - no hurry - helped me in this weekend with this workaround 
if i put the bt.c in the project folder then where i work, the no stop is there in bt.c

perhabs a path mistake in a lib? config?

hope you guys are fine.
and take care of yourself - you are wanted too ;-) 
https://twitter.com/eMbeddedHome/status/778997188978442240

best wishes
rudi ;-)

igrr, 2016-09-24:
I don't have a Windows VM at hand, my guess is that `./` may be causing problems.
Try removing this line: https://github.com/espressif/esp-idf/blob/master/components/bt/component.mk#L23
It isn't really necessary because by default the build system will compile all files in component root directory.

As far as I can tell, includes are in order, but if you get compilation error, remove the `./` from include line as well.

ESP32DE, 2016-09-24:
yes ;-) this did the trick in windows / v7 
- hello world
- bt app
without stop/mistake now done.

usually use linux too, but the used ide is a win app.

txs ivan
best wishes
rudi ;-)

to do in windows like you point

# comment this line in windows environment  
# COMPONENT_SRCDIRS := ./

( can be closed - not sure - can i close now ? will wait for one day - then i close or the stuff then close )

igrr, 2016-09-24:
Great to know that it works now. You can keep this issue open, I'll close it when the fix is pushed to GitHub.

igrr, 2016-09-26:
Should be fixed in 10c6951.

closed 2016-09-26 by igrr