espressif/esp-idf #23: Can not get ap list when calling esp_wifi_get_ap_list()

comcat, 2016-09-24:
Seems like the type of second parameter of esp_wifi_get_ap_list should be (wifi_ap_list_t **)

ESP32DE, 2016-09-24:
not sure perhabs helps

esp_err_t esp_wifi_get_ap_list(uint16_t *number, wifi_ap_list_t *ap_list);

typedef enum {
    WIFI_PS_NONE,    /**< No power save */
    WIFI_PS_MODEM,   /**< Modem power save */
    WIFI_PS_LIGHT,   /**< Light power save */
    WIFI_PS_MAC,     /**< MAC power save */
} wifi_ps_type_t;

ESP32DE, 2016-09-24:
not sure perhabs helps:

`typedef struct {
    uint8_t bssid[6];                     /**< MAC address of AP */
    uint8_t ssid[32];                     /**< SSID of AP */
    uint8_t primary;                      /**< channel of AP */
    wifi_second_chan_t second;            /**< second channel of AP */
    int8_t  rssi;                         /**< signal strength of AP */
    wifi_auth_mode_t authmode;            /**< authmode of AP */
} wifi_ap_list_t;

**
  * @brief     Get AP list found in last scan
  *
  * @param     uint16_t *number : as input param, it stores max AP number ap_list can hold, as output param, it store
                                  the actual AP number this API returns
  * @param     wifi_ap_list_t *ap_list : a list to hold the found APs
  *
  * @return    ESP_OK : succeed
  * @return    others : fail
  */
esp_err_t esp_wifi_get_ap_list(uint16_t *number, wifi_ap_list_t *ap_list);

`

comcat, 2016-09-24:
'wifi_ap_list_t *' should not get the returned pointer of array in C

nkolban, 2016-09-25:
I would imagine that to use the API called esp_wifi_get_ap_list() one allocates enough storage to hold the maximum number of access point structures you are willing to retrieve ...  for example ... 10.  You then call esp_wifi_get_ap_list() passing in 10 (the number of elements you have allocated storage for) and a pointer to a contiguous chunk of storage that is large enough for 10 elements.  On return, you are told the number of elements that have been filled in.  The storage pointed to by the pointer you passed in has thus been updated to hold that many ap entries.

negativekelvin, 2016-09-25:
Well you can get the number of results before you call esp_wifi_get_station_list with esp_wifi_get_ap_num.  But I wonder are the results returned by esp_wifi_get_station_list sorted by rssi?  So if you only request 5 wifi_ap_list_t results they will have the highest rssi.

Spritetm, 2016-09-25:
My guess would be that someone indeed forgot an extra * somewhere. We will look into this after the weekend.

igrr, 2016-09-25:
Actually, seems to work okay here.
Here's example code, we will add it to the examples directory soon:

```c++
void scan_done_handler(void)
{
    uint16_t sta_number;
    esp_wifi_get_ap_num(&sta_number); // get the number of scanned APs
    wifi_ap_list_t* ap_list_buffer = malloc(sta_number * sizeof(wifi_ap_list_t));
    if (ap_list_buffer == NULL){
        return;
    }

    if (esp_wifi_get_ap_list(&sta_number, ap_list_buffer) == ESP_OK) {
        for(int i=0; i<sta_number; i++) {
              // do something with ap_list_buffer[i]
        }
    }
    free(ap_list_buffer);
}
```

comcat, 2016-09-26:
It's OK. Thx.

closed 2016-09-26 by comcat