espressif/esp-idf #24: Suggestion: Better name for the wifi_ap_list_t data type

nkolban, 2016-09-25:
When working with access points, we can call `esp_wifi_get_ap_list()` to retrieve the discovered access points.  The input to this function is a contiguous memory area that will be populated with records corresponding to each discovered access point.  In other words, it is an array of access point data structures.  However, the data type that has been chosen for a single access point is called `wifi_ap_list_t` which makes one think that a record is itself a whole list ... when a record is ... well ... a record.

I'd like to suggest renaming this data type from:

`wifi_ap_list_t`

to something else.  Suggestions might include:

* `wifi_ap_record_t`
* `wifi_ap_entry_t`
* `wifi_ap_detail_t`

negativekelvin, 2016-09-26:
Or wifi_ap_scan_result_t ?

nkolban, 2016-09-26:
That would be good too ... as long as the data structure serves no other purpose than holding a scan result.

bvernoux, 2016-10-07:
I think the best will be wifi_ap_record_t

igrr, 2016-10-27:
Renamed to `wifi_ap_record_t` (but possibly will be renamed again to `esp_wifi_ap_record_t` in a few days).

closed 2016-10-27 by igrr