espressif/esp-idf #25: make menuconfig not work on macos sierra

widora, 2016-09-26:
### the pre-installed package is below:
$ brew install gnu-sed gawk binutils gperf grep gettext ncurses
Warning: gnu-sed-4.2.2 already installed
Warning: gawk-4.1.3_1 already installed
Warning: binutils-2.27 already installed
Warning: homebrew/dupes/gperf-3.0.4 already installed
Warning: homebrew/dupes/grep-2.25 already installed
Warning: gettext-0.19.8.1 already installed
Warning: homebrew/dupes/ncurses-6.0_1 already installed
Warning: You are using OS X 10.12.
We do not provide support for this pre-release version.
make menuconfig error: can't found ncurses!
### $ make menuconfig
MAKEFLAGS="" \
	CC=cc LD=ld \
	/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/mango/ESP32/IDF/esp-idf/tools/kconfig
lxdialog/check-lxdialog.sh -check cc  -I/usr/local/opt/ncurses/include/ncurses -DCURSES_LOC="<ncurses.h>" -DLOCALE -lintl 
 *** Unable to find the ncurses libraries or the
 *** required header files.
 *** 'make menuconfig' requires the ncurses libraries.
 *** Install ncurses (ncurses-devel) and try again.
make[1]: *** [dochecklxdialog] Error 1
make: *** [/Users/mango/ESP32/IDF/esp-idf/tools/kconfig/mconf] Error 2

ritazh, 2016-09-26:
+1 

I also got errors trying to run `make menuconfig` on macos sierra

Following the [instructions](https://github.com/espressif/esp-idf/blob/master/docs/macos-setup.rst) for Mac. After running `make menuconfig`, I got the follow error: 

```
$ make menuconfig
MAKEFLAGS="" \
	CC=cc LD=ld \
	/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/ritazhang/Documents/work/esp/esp32/esp-idf/tools/kconfig
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o mconf.o mconf.c
In file included from mconf.c:22:
./lkc.h:12:11: fatal error: 'libintl.h' file not found
# include <libintl.h>
          ^
1 error generated.
make[1]: *** [mconf.o] Error 1
make: *** [/Users/ritazhang/Documents/work/esp/esp32/esp-idf/tools/kconfig/mconf] Error 2
```
Then, I ran the following. 

```
$ sudo port install gettext
$ sudo port install php55-gettext
$ brew unlink gettext && brew link gettext --force
```
Following `make menuconfig`, got these error:

```
$ make menuconfig
MAKEFLAGS="" \
	CC=cc LD=ld \
	/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/ritazhang/Documents/work/esp/esp32/esp-idf/tools/kconfig
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE -lintl 
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lintl 
Undefined symbols for architecture x86_64:
  "_acs_map", referenced from:
      _dialog_checklist in checklist.o
      _print_arrows in checklist.o
      _dialog_clear in util.o
      _draw_box in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      ...
  "_cbreak", referenced from:
      _init_dialog in util.o
  "_delwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_doupdate", referenced from:
      _dialog_checklist in checklist.o
  "_endwin", referenced from:
      _init_dialog in util.o
[... 136 lines cut ...]
      _dialog_textbox in textbox.o
      _print_page in textbox.o
      _dialog_menu in menubox.o
  "_wprintw", referenced from:
      _print_item in checklist.o
      _print_position in textbox.o
  "_wrefresh", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_buttons in checklist.o
      _end_dialog in util.o
      _dialog_inputbox in inputbox.o
      _print_buttons in inputbox.o
      _refresh_text_box in textbox.o
      ...
  "_wscrl", referenced from:
      _dialog_checklist in checklist.o
      _do_scroll in menubox.o
  "_wtouchln", referenced from:
      _attr_clear in util.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [mconf] Error 1
make: *** [/Users/ritazhang/Documents/work/esp/esp32/esp-idf/tools/kconfig/mconf] Error 2
```

widora, 2016-09-26:
@ritazh libintl.h
add #define KBUILD_NO_NLS in lkc.h,line num 10:
``` bash
  1 /*
  2  * Copyright (C) 2002 Roman Zippel <zippel@linux-m68k.org>
  3  * Released under the terms of the GNU GPL v2.0.
  4  */
  5 
  6 #ifndef LKC_H
  7 #define LKC_H
  8 
  9 #include "expr.h"
 10 #define KBUILD_NO_NLS
 11 #ifndef KBUILD_NO_NLS
 12 # include <libintl.h>
 13 #else
```
will be compile OK,the follow error is ncurses

lakidd, 2016-09-26:
+1 I had the same issue on El Cap.
I'm not sure exactly what fixed it but it was either
a) going back to bash from zsh
or 
b) installing ncurses6.0 from source
or a combination of both

igrr, 2016-09-26:
Thanks for all the reports. Installing clean OS X VM here, will see what's missing.

widora, 2016-09-26:
Once I compile OpenWrt, its menuconfig is also based on ncurses, no problem. so esp-idf shouldn't be a problem

igrr, 2016-09-26:
Okay, with a clean install of homebrew on OS X 10.11. Found two ways to do this.

First one, dirty:
- `brew link gettext --force && brew link ncursesw --force`
- Then `make` worked fine.

Second one, hacky:
- `brew install pkg-config`, `export PKG_CONFIG_PATH=/usr/local/Cellar/ncurses/6.0_2/lib/pkgconfig`
- homebrew gettext package doesn't have pkgconfig files, so I had to manually add paths to gettext paths to check-lxdialog.sh:
- patch `tools/kconfig/lxdialog/check-lxdialog.sh`:
```diff
diff --git a/tools/kconfig/lxdialog/check-lxdialog.sh b/tools/kconfig/lxdialog/check-lxdialog.sh
index 6964d2b..877d571 100755
--- a/tools/kconfig/lxdialog/check-lxdialog.sh
+++ b/tools/kconfig/lxdialog/check-lxdialog.sh
@@ -6,7 +6,7 @@ ldflags()
 {
        if [ $(uname -s) == "Darwin" ]; then 
                #OSX seems to need intl too
-               echo -n "-lintl "
+               echo -n "-lintl -L/usr/local/Cellar/gettext/0.19.8.1/lib "
        fi
        pkg-config --libs ncursesw 2>/dev/null && exit
        pkg-config --libs ncurses 2>/dev/null && exit
@@ -25,6 +25,7 @@ ldflags()
 # Where is ncurses.h?
 ccflags()
 {
+       echo '-I/usr/local/Cellar/gettext/0.19.8.1/include'
        if pkg-config --cflags ncursesw 2>/dev/null; then
                echo '-DCURSES_LOC="<ncurses.h>" -DNCURSES_WIDECHAR=1'
        elif pkg-config --cflags ncurses 2>/dev/null; then
```
4. then `make` worked fine

Haven't tried 10.12 yet (still downloading) and a clean MacPorts install (still downloading Xcode).
Will update this issue once i know more.

ricardoquesada, 2016-09-27:
I fixed (hardcode the fix) with this commit:
https://github.com/ricardoquesada/esp-idf/commit/081de7967e78f4a5255979fc21f89f152e8176db

igrr, 2016-09-27:
Tested with a clean install of macports on OS X 10.11.
Only `sudo port install gsed gawk binutils gperf grep gettext ncurses pkgconfig` is needed, then `make` works fine.

ricardoquesada, 2016-09-28:
was this issue closed because it was fixed? or because it works with macport?
the issue was opened because it doesn't work with `brew` (and not macport).

NAzT, 2016-10-04:
I'm using `brew` and fixed this issue the error by add PKG_CONFIG to my .profile

export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH";

[image]

reference to this thread:
https://www.facebook.com/groups/ChiangMaiMakerClub/permalink/1072272859559473/?comment_id=1073069399479819&comment_tracking=%7B%22tn%22%3A%22R%22%7D

FYI, I've install following packages using homebrew.

$ brew tap homebrew/dupes
$ brew install binutils coreutils automake wget gawk libtool help2man gperf gnu-sed --with-default-names grep
$ export PATH="/usr/local/opt/gnu-sed/libexec/gnubin:$PATH"

ricardoquesada, 2016-10-04:
I confirm that @NAzT works.

lakidd, 2016-10-05:
+1 @NAzT with zsh

wolfeidau, 2016-10-19:
Yeah I got this error and had to run `brew link gettext --force && brew link ncursesw --force` to get it working.

I am on 10.11.6 El Capitano.

Cheers

MobiusHorizons, 2016-11-04:
still seeing the ncurses error on Sierra, even though i have installed and force linked `gettext` and `ncurses` on my system. I should note that i could not find a package `ncursesw` if that is important.

MobiusHorizons, 2016-11-05:
Never mind, @NAzT's solution worked.

igrr, 2016-11-07:
Should be fixed in latest master.

closed 2016-11-07 by igrr

ttimasdf, 2017-10-26:
Got the very same error as @ritazh while `make menuconfig` in Android kernel, and google leads me to this page. 

As a note for others who googled like me:
Just invoke `make nconfig` instead do solve the problem, it has a similar UI compared with menuconfig. no need to look through and hack around.