espressif/esp-idf #26: Unable to make flash

ritazh, 2016-09-26:
Following the [instructions](https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst) for linux. After running `'make menuconfig`, set the hardware to Download mode, then I ran `make flash`, here is what I’m getting. Note: This is a linux VM running on MacOS host using Parellels. Mapped host port `/dev/tty.SLAB_USBtoUART` to vm port `/dev/ttylS0`. 

Flashing project app to 0x10000...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Calling 4009068c
Stub running...
Attaching SPI flash...
Configuring flash size...
Erasing flash...
Took 0.01s to erase flash block
Wrote 8192 bytes at 0x00001000 in 0.7 seconds (89.4 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.01s to erase flash block
Writing at 0x00010000... (1 %) 
A fatal error occurred: Invalid head of packet ('\x00')
/home/ritazhang/Desktop/esp/esp-idf/components/esptool_py/Makefile.projbuild:27: recipe for target 'flash' failed
make: *** [flash] Error 2

Spritetm, 2016-09-26:
This sometimes happens on a non-VM as well if the baudrate selected is too high for the cabling/adapter/driver/... Can you try running a make menuconfig and then under 'Serial flasher config', select a lower default baudrate? 115200 should normally work unless you have some very broken hardware.

ritazh, 2016-09-26:
Thanks for the suggestion @Spritetm. I just tried a lower baud rate: 74880. Got it to flash but stopped at 56%. See log below.

@wujiangang FYI.

```
Flashing project app to 0x10000...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Calling 4009068c
Stub running...
Attaching SPI flash...
Configuring flash size...
Erasing flash...
Took 0.01s to erase flash block
Wrote 8192 bytes at 0x00001000 in 1.4 seconds (45.2 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.01s to erase flash block
Writing at 0x0004c000... (56 %) 
A fatal error occurred: Invalid head of packet ('\x00')
/home/ritazhang/Desktop/esp/esp-idf/components/esptool_py/Makefile.projbuild:27: recipe for target 'flash' failed
make: *** [flash] Error 2
```

projectgus, 2016-09-26:
Hi @ritazh,

Thanks for clarifying and providing the extra debug dump.

If you connect the USB device to the VM's USB controller, rather than passing through the serial port (so it appears as /dev/ttyUSBx rather than /dev/ttylsx), does this make a difference?

Otherwise, to second what @Spritetm said, is there anything unsual about your ESP32 hardware or the way it is connected or powered? Errors like the one you're seeing can sometimes be caused by power supply or serial cabling issues, if it's not the VM.

ritazh, 2016-09-27:
Hi @projectgus Thanks for the USB suggestion. I think we are getting closer. After I mapped the serial port AND the USB port to the Linux VM, then `make menuconfig` now uses `/dev/ttyUSBx` instead of `/dev/ttySx`, I see the flash going all the way to 100%. But still getting the following error. I did press the `Boot` button, then the `EN` button before running `make flash` to enter download mode. The hardware I have is `ESP-c-0022`.

```
make[1]: Entering directory '/home/ritazhang/Desktop/esp/esp-idf/components/bootloader/src'
make[2]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/bootloader/log'
make[2]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/bootloader/log'
make[2]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/bootloader/main'
make[2]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/bootloader/main'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/esp-idf/components/bootloader/src'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/bt'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/bt'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/driver'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/driver'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/esp32'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/esp32'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/expat'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/expat'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/freertos'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/freertos'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/json'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/json'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/log'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/log'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/lwip'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/lwip'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/mbedtls'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/mbedtls'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/nvs_flash'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/nvs_flash'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/spi_flash'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/spi_flash'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/tcpip_adapter'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/tcpip_adapter'
make[1]: Entering directory '/home/ritazhang/Desktop/esp/myapp/build/main'
make[1]: Leaving directory '/home/ritazhang/Desktop/esp/myapp/build/main'
Flashing project app to 0x10000...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Calling 4009068c
Stub running...
Attaching SPI flash...
Configuring flash size...
Erasing flash...
Took 0.01s to erase flash block
Wrote 8192 bytes at 0x00001000 in 1.1 seconds (60.7 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.01s to erase flash block
Wrote 450560 bytes at 0x00010000 in 59.5 seconds (60.6 kbit/s)...
Hash of data verified.
Erasing flash...
Took 0.01s to erase flash block
Wrote 8192 bytes at 0x00004000 in 1.1 seconds (61.1 kbit/s)...
Hash of data verified.

Leaving...

A fatal error occurred: Failed to enter RAM download mode (result was 0xff, 0x0)
/home/ritazhang/Desktop/esp/esp-idf/components/esptool_py/Makefile.projbuild:27: recipe for target 'flash' failed
make: *** [flash] Error 2
```

Spritetm, 2016-09-27:
That is a known issue in the current esptool.py version we use. The error is harmless and can be ignored; we have an esptool.py submodule update in the queue that should get rid of it completely.

ritazh, 2016-09-27:
@Spritetm Awesome! Just got an ip from the sample program running on the board! Thanks for the help everyone!!! 👍 🎁

projectgus, 2016-10-04:
HI @ritazh ,

[Great ESP32 getting started guide!](https://ritazh.com/get-started-with-esp32-785a111476f2) I'm guessing we can close this issue now.

closed 2016-10-04 by projectgus