espressif/esp-idf #27: examples/03_http_request crashed

comcat, 2016-09-27:
Only config the ssid & password and disable the 'CONFIG_ESPTOOLPY_COMPRESSED' through 'make menuconfig'

Boot log:
-------------------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:920
load:0x40078000,len:2624
load:0x40098000,len:508
entry 0x40098118
I (93) heap_alloc_caps: Initializing heap allocator:
I (93) heap_alloc_caps: Region 19: 3FFC07DC len 0001F824 tag 0
I (94) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (104) cpu_start: Pro cpu up.
I (109) cpu_start: Starting app cpu, entry point is 0x400809e4
I (0) cpu_start: App cpu up.
I (125) cpu_start: Pro cpu start user code
rtc v112 Sep 22 2016 16:08:39
XTAL 40M
I (161) cpu_start: Starting scheduler on PRO CPU.
I (43) cpu_start: Starting scheduler on APP CPU.
misc_nvs_load g_misc_nvs=0x3ffc493c
frc2_timer_task_hdl:3ffc53d8, prio:22, stack:2048
tcpip_task_hdlxxx : 3ffc5f3c, prio:18,stack:2048
phy_version: 123, Sep 13 2016, 20:01:58, 0
pp l
pp_task_hdl : 3ffc9798, prio:23, stack:8192
I (3413) example: Setting WiFi configuration SSID GAGA-comcat...
mode : sta(18:fe:34:6a:87:88)
Guru Meditation Error of type LoadStoreError occurred on core   0. Exception was unhandled.
Register dump:
PC      :  4008471d  PS      :  00060733  A0      :  80082024  A1      :  3ffc9570  
A2      :  400d5100  A3      :  b33fffff  A4      :  b33f0000  A5      :  00060723  
A6      :  b33fffff  A7      :  00000000  A8      :  800820c8  A9      :  3ffc9570  
A10     :  00000003  A11     :  00060723  A12     :  00060723  A13     :  00000000  
A14     :  3ffc9b5c  A15     :  3ffc9b24  SAR     :  00000018  EXCCAUSE:  00000003  
EXCVADDR:  400d5100  LBEG    :  4000c349  LEND    :  4000c36b  LCOUNT  :  ffffffff  
Rebooting...
---------------------------------------------

objdumped the 0x4008471d:
----------------------------------------
.......
40084714 <uxPortCompareSet>:
40084714:   004136          entry   a1, 32
40084717:   130c30          wsr.scompare1   a3
4008471a:   002000          isync
4008471d:   00e242          s32c1i  a4, a2, 0            <---- access the address of code (a2 = 400d5100)
40084720:   042d        mov.n   a2, a4
40084722:   f01d        retw.n
......

negativekelvin, 2016-09-28:
Make sure you updated to the latest WiFi libraries?

comcat, 2016-09-28:
The http example is OK after updating the esp32-wifi-lib by hand. Close.

closed 2016-09-28 by comcat