espressif/esp-idf #29: BT scanning issues

jmattsson, 2016-09-28:
I've been experimenting with the bluetooth stack today and got a NodeMCU-based proof-of-concept going with advertisements relatively easily (have the example as reference was quite useful, thanks!). However, I'm hitting a wall when it comes to scanning on the ESP32.

  1. While it reports that the scan enable command was successful, I never receive any advertising reports, despite there being plenty of advertising BT devices around. I've tried explicitly setting the event mask to include advertising reports, but that did not help (and the default value should be enough anyway). Other than setting the parameters and then enabling the scan, am I supposed to do something else?
  1. When an LE SET SCAN ENABLE command has been sent once (whether to enable or disable), all subsequent LE SET SCAN ENABLE commands report failure, code 0x0c (Command Disallowed). As such, it does not appear to be possible to stop scanning other than via a HCI_RESET command. **Edit:** Actually, I see the same behaviour with LE SET ADVERTISE ENABLE as well; once on, it's not going off.

Are these bugs, or am I just doing things wrong? I haven't got the code checked in or pushed anywhere yet (it needs a lot of cleanup, and ideally it should also work...), but here is some example output:
```lua
> bt.reset()
BT: 04 0e 04 05 03 0c 00
BT command 0c03 completed, 5 queue free, retval(s):  00

> bt.scan.setparams({mode=1,interval=32,window=16,own_addr_type=0,filter_policy=0})
BT: 04 0e 04 05 0b 20 00
BT command 200b completed, 5 queue free, retval(s):  00

> bt.scan.enable(1)
BT: 04 0e 04 05 0c 20 00
BT command 200c completed, 5 queue free, retval(s):  00

> -- No advertising events generated :(
>
> bt.scan.enable(0)
BT: 04 0e 04 05 0c 20 0c
BT command 200c completed, 5 queue free, retval(s):  0c
> -- and I can't even stop the scanning!
```
And for the advertising:
```lua
> bt.reset()
BT: 04 0e 04 05 03 0c 00
BT command 0c03 completed, 5 queue free, retval(s):  00

> bt.adv.setparams({type=bt.adv.NONCONN_UNDIR})
BT: 04 0e 04 05 06 20 00
BT command 2006 completed, 5 queue free, retval(s):  00

> bt.adv.setdata(encoder.fromHex("080861626364656667"))
BT: 04 0e 04 05 08 20 00
BT command 2008 completed, 5 queue free, retval(s):  00

>bt.adv.enable(1)
BT: 04 0e 04 05 0a 20 00
BT command 200a completed, 5 queue free, retval(s):  00

> -- happily advertising at this point
>
> bt.adv.enable(0)
BT: 04 0e 04 05 0a 20 0c
BT command 200a completed, 5 queue free, retval(s):  0c

> -- and will be forever more, because the stop command gets rejected
```

Also, currently the BT stack uses callbacks for delivering events/responses. Which task context are those executed from? And is the BT stack expected to move over to use `esp_event_send()` like the WiFi stack in the near future?

TianHao-Espressif, 2016-09-28:
Hi, 
    Thank you for paying attention to bluetooth reference.
    You said you have sent "event_mask", besides, do you have  sent "set_event_mask_enable" hci command ? 
    And, this VHCI APIs is for common hci interface, so it doesn't have task conext. If you receive something , you must do xQueueSend to switch to your application task.

jmattsson, 2016-09-28:
A-ha! I'd missed the LE Meta-Event bit in the Controller/Baseband event mask. I'm now seeing the advertisements, thanks!

I still can't stop the scan (or advertising) without a full HCI reset however.

TianHao-Espressif, 2016-09-28:
Hi，
  We do the experiment to try stop the advertising stop, it can be worked. So, could you check your hci command content? Or you send advertising stop too quickly after send advertising enable?

negativekelvin, 2016-09-28:
@TianHao-Espressif  hi, which Bluetooth protocol stack will be used in esp-idf?  Coming soon?

TianHao-Espressif, 2016-09-28:
@negativekelvin  Hi, ble reference will be coming firstly.

TianHao-Espressif, 2016-09-28:
@negativekelvin , actually, we will support classic bt and ble dual mode.

negativekelvin, 2016-09-28:
@TianHao-Espressif OK thanks, but will protocol stack be full open source above hci layer?  Who is stack provider?

TianHao-Espressif, 2016-09-29:
@negativekelvin ， yeah，the protocol will be open source above hci layer.

jmattsson, 2016-09-29:
@TianHao-Espressif I found a bug in my code that was causing the issue with stopping advertising and scanning. Apologies for the noise - everything is now working. Thanks for your help!

closed 2016-09-29 by jmattsson

RSchlenker, 2016-11-24:
Hi jmattson,
I am currently struggling with the bluetooth part for esp32.
I am not sure where to start here, since it's my first bluetooth esp project. I would be very happy to see your bluetooth code on github, to have a small reference. Even if the code isn't too clean.

Is there any BLE-API Reference beside the example code and the address definitions in "/components/esp32/ld/esp32.rom.ld". (I just tried to understand the ble example)
Kind regards

jmattsson, 2016-11-24:
I haven't got much beyond the example code myself; we're still waiting for Espressif to release the full BT LE & Classic stacks. Right now it's only the VHCI that's available for use, the higher level protocols aren't there yet. As for code, it's all in the NodeMCU32 [bthci](https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/components/modules/bthci.c) module, which has its Lua interface documented [here](https://github.com/nodemcu/nodemcu-firmware/blob/dev-esp32/docs/en/modules/bthci.md).

RSchlenker, 2016-11-24:
Ok, thank you!

diegojpalaquibay, 2017-09-04:
@TianHao-Espressif, Hello, I am doing a continuous  BLE scanner and it works well for a couple minutes, then suddenly the ble scanner doesnt detect any ble advertisement, and after a couple of minutes it detects again, do U have some advice to make that scanner detect the ble advertisement all the time? I am using the gatt_client example to make my scanner
 Regards
Diego

TianHao-Espressif, 2017-09-04:
@diegojpalaquibay ,  Thank you pay attention on ESP32. We're debugging on this scan problem. Please wait for a little time.

diegojpalaquibay, 2017-09-04:
@TianHao-Espressif , thank you very much for your help, please let me know when u get  some news for this problem...thx 
Regards
Diego

sharkyy, 2018-03-17:
Got the same problem, is there a workaround available for this?

Thanks and regards
Max