espressif/esp-idf #30: Windows 10: use msys2_shell download  error

eboxmaker, 2016-09-29:
Flashing project app to 0x10000...
esptool.py v2.0-dev
Traceback (most recent call last):
  File "C:/esp32_idf/esp-idf/components/esptool_py/esptool/esptool.py", line 1813, in <module>
    main()
  File "C:/esp32_idf/esp-idf/components/esptool_py/esptool/esptool.py", line 1642, in main
    esp = chip_constructor_fun(args.port, initial_baud)
  File "C:/esp32_idf/esp-idf/components/esptool_py/esptool/esptool.py", line 159, in __init__
    self._port = serial.Serial(port)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialutil.py", line 182, in __init__
    self.open()
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/??/COM3': WindowsError(3, '\xcf\xb5\xcd\xb3\xd5\xd2\xb2\xbb\xb5\xbd\xd6\xb8\xb6\xa8\xb5\xc4\xc2\xb7\xbe\xb6\xa1\xa3')
make: *** [C:/esp32_idf/esp-idf/components/esptool_py/Makefile.projbuild:28：flash] 错误 1

**could not open port '/??/COM3':**  What I can do to solve this problem ?

negativekelvin, 2016-09-29:
Is your port defined in menuconfig as COM3 and not the msys port /dev/ttyS2?

eboxmaker, 2016-09-29:
yes，you are right，Modify /dev/ttyS2 to / COM3 just fine.
thank you。

closed 2016-10-10 by eboxmaker