espressif/esp-idf #31: esp-idf kconfig build error on macOS (Sierra) Undefined symbols for architecture x86_64

NAzT, 2016-09-29:
I got error: ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
while compile my component here's the full logs

https://gist.github.com/NAzT/b9c9a02cd17abcfbf3fb6bffc6d59aa0

I've compiled esp32 toolchain on this machine without any errors.

here's an error
[image]

compiling toolchain 
[image]

closed 2016-10-01 by NAzT

projectgus, 2016-10-03:
Hi @NAzT,

Hoping you closed this issue because you figured out the problem. Are there any tips you can give for someone who finds this Issue in the future, or anything we can add to esp-idf?

Thanks,

Angus

NAzT, 2016-10-04:
@projectgus here's the details:

I'm using brew and fixed this issue the error by add PKG_CONFIG_PATH to my .profile

export LIBRARY_PATH="/usr/local/opt/gettext/lib:$LIBRARY_PATH";
export PKG_CONFIG_PATH="/usr/local/opt/ncurses/lib/pkgconfig:$PKG_CONFIG_PATH"
export CPATH="/usr/local/opt/gettext/include:$CPATH";

screen shot 2559-10-04 at 3 26 18 pm

reference to this thread:
https://www.facebook.com/groups/ChiangMaiMakerClub/permalink/1072272859559473/?comment_id=1073069399479819&comment_tracking=%7B%22tn%22%3A%22R%22%7D

FYI, I've install following packages using homebrew.

$ brew tap homebrew/dupes
$ brew install binutils coreutils automake wget gawk libtool help2man gperf gnu-sed --with-default-names grep
$ export PATH="/usr/local/opt/gnu-sed/libexec/gnubin:$PATH"