espressif/esp-idf #35: Eclipse on Windows: Symbol 'CONFIG_BLINK_GPIO' could not be resolved

krzychb, 2016-09-30:
Hi All,

I am unable to get rid of errors below. 
As project build is successful, I guess the issue is similar to #17.

[image]

Are you aware of any workaround to Eclipse configuration (like proposed in #17), so such renaissance errors are not reported?

negativekelvin, 2016-10-02:
Did you run make menuconfig first?

krzychb, 2016-10-02:
@negativekelvin, 

Yes, this is after I run `make menuconfig` first. 

I have noticed that such bogus errors may disappear after saving the main.c file and building the project again. Then then return back after the following build. So far I was unable to find a workflow to reliably get rid of them. This issue concerns other examples as well and other variables like `portTICK_RATE_MS`. 

The other set  up I made using Ubuntu 16.04 LTS does not have any of such issues at all.

projectgus, 2016-10-03:
Hi @krzychb,

Thanks for reporting this. You're correct that it's the same issue as #17, Eclipse isn't parsing all of the include paths from the esp-idf build process correctly on Windows (even though the build is succeeding in the background.)

I'm going to close this issue in favour of #17, but please subscribe to that issue to get updates on a fix.

There is a workaround (mentioned in the other issue) of manually adding include paths to the project configuration so Eclipse finds the header files every time. We're aiming to have a proper fix soon.

Angus

closed 2016-10-03 by projectgus