espressif/esp-idf #36: Poorly formatted message for stack overflow for task

nkolban, 2016-10-01:
In an application, I coded the following:

```
xTaskCreatePinnedToCore(&espruinoTask, "espruinoTask", 2048, NULL, 5, NULL, 0);
```
It seems that I ran out of stack space ... however the error that was logged was:

```
***ERROR*** A stack overflow in taskespruinoTaskhas been detected.
```

I think a better formatted message would be:
```
***ERROR*** A stack overflow in task espruinoTask has been detected.
```
(notice the extra spaces)

ESP32DE, 2016-10-01:
good catch ..
your code
 # xTaskCreatePinnedToCore(&espruinoTask, "espruinoTask", 2048, NULL, 5, NULL, 0);
you named the Task : espruinoTask

there is no wrong with the message
 # ***ERROR*** A stack overflow in taskespruinoTaskhas been detected.
 but the missing space between your Task Name and the error entry Name "Task" is right.
so here the error must insert a space.

but .. 
but your suggest:
# ***ERROR*** A stack overflow in task espruinoTask has been detected.

is now wrong  ;-)
you did insert now in the  Name of your Task a space between task espruino and append the entry Name "Task" on your Task Name.

;-)

projectgus, 2016-10-04:
Thanks @nkolban & @ESP32DE. We have this fix in our internal review queue for merging.

commit 71c09d8f (2016-10-13): Fix stack overflow message format

Poorly formatted message for stack overflow for task. Closes #36
closed 2016-10-13 by igrr