espressif/esp-idf #38: esp32_out.ld dependency breaks when BUILD_DIR_BASE contains subdir

jmattsson, 2016-10-03:
In the NodeMCU CI we use the documented BUILD_DIR_BASE to build different configurations into different output directories (e.g. `build/float` and `build/integer`). With the latest IDF the hardcoded dependency on `../include/sdkconfig.h` breaks, as it would now be ../../include/sdkconfig. That is, this line:
`esp32_out.ld: $(COMPONENT_PATH)/ld/esp32.ld ../include/sdkconfig.h`

More things break if BUILD_DIR_BASE is set to something that isn't part of `build/` as Kconfig appears to ignore BUILD_DIR_BASE and insists on writing to `build/` even if BUILD_DIR_BASE is set to e.g. `/tmp/mybuild`.

I believe the short-term fix would be to add a VPATH (or the lower-case version thereof) for `$(PROJECT_PATH)/build/include`, and list the dependency as simply `sdkconfig.h`.

projectgus, 2016-10-04:
Hi @jmattsson ,

Thanks for reporting this. It looks like there were a few problems preventing building out-of-tree via `BUILD_DIR_BASE`.

I've submitted fixes into our internal test/review system and also added out-of-tree test cases to the automatic build system tests. It might be a little while before those fixes get merged to master, as China is on holidays this week (Golden Week).

So in the meantime I've also pushed the fixes here:
https://github.com/espressif/esp-idf/tree/bugfix/build_dir_base

Please let me know if it fixes this issue.

Angus

jmattsson, 2016-10-04:
Thanks, it looks like you've done a thorough sweep here! I'll try to give this a spin tomorrow.

jmattsson, 2016-10-05:
Mixed results:

*  Out-of-tree build worked, but yielded a whole bunch of:
  `make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.`
  On this branch I also get these when not overriding BUILD_DIR_BASE. The master branch does not   have this issue. I build with `-j 8 -l 8` in MAKEFLAGS by default.

* In-tree build with `BUILD_DIR_BASE=build/integer` fails with:
  `make[2]: *** No rule to make target 'build/float/bootloader/bootloader.bin'.  Stop.`

projectgus, 2016-10-05:
> warning: jobserver unavailable: using -j1. Add '+' to parent make rule.

Interesting, will investigate. We clear MAKEFLAGS when we recurse a couple of times, to avoid other make-related weirdness. Possibly we weren't actually clearing MAKEFLAGS correctly one of those times!

> In-tree build with BUILD_DIR_BASE=build/float fails

Ah, I missed that nuance in the original bug report. Will add support for providing a relative path in this variable.

jmattsson, 2016-10-05:
Parallel builds have been working great so far, I haven't run into any dependency weirdness (that I've noticed).

projectgus, 2016-10-06:
The problem wasn't parallel-build related, it was to do with setting other MAKEFLAGS that we didn't want propagating to some submakes.

I think I've taken care of it now. No more jobserver warnings, and I can build when BUILD_DIR_BASE is a relative path. Please let me know how you go!

jmattsson, 2016-10-06:
Looking good here too, thanks!

commit 66882347 (2016-10-20): build system: Fix out-of-tree building via BUILD_DIR_BASE

Closes #38
closed 2016-10-20 by igrr