espressif/esp-idf #39: Defaut *.bin files from manufacturing ? If needed to reprogram them ...

JPB1JPB1, 2016-10-03:
Hello,
can you provide the defaut *.bin file from manufacturing or shipping if they are needed to reprogram the ESP32.
At the same, can you read only the ESP-32 and store the bin in new using the Flash tool software.
So that we can program, read back in file and do a binary compare of all the files...

Thanks and great ESP32.
Jp

projectgus, 2016-10-04:
Hi @JPB1JPB1,

> can you provide the defaut *.bin file from manufacturing or shipping if they are needed to reprogram the ESP32.

Because the ESP32 chip itself doesn't have any on-board flash, we (Espressif) don't ship any factory image with the chip.

If the ESP32 comes in a module (like ESP-WROOM-32 or ESP3212) then there is flash onboard and it can be pre-loaded with something from the factory.

The module ESP-WROOM-32 is manufactured by us, I'll double check if any image comes pre-flashed onboard but I don't believe it does.

Other modules like ESP3212 are flashed with firmware by the module manufacturer (Ai-Thinker in the case of ESP3212), not by us.

> At the same, can you read only the ESP-32 and store the bin in new using the Flash tool software.

You can use esptool.py (included in esp-idf) to save and restore flash from the command line:

For a 4MB flash chip (0x400000 bytes), you can run something like this in the esp-idf directory (using the same terminal you use to compile your esp-idf project, with IDF_PATH environment variable set as per the instructions):

Reset the module to bootloader mode, same as for regular esp-idf flashing. Then run:

```
source ${IDF_PATH}/add_path.sh
esptool.py --baud 920600 read_flash 0 0x400000 original_contents.bin
```

* `source add_path.sh` extends your PATH to include th path to some tools shipped with esp-idf, including esptool.py.
* read_flash reads the full 4MB of flash contents to a file. You will need a different (smaller) number if your module shipped with less than 4MB of flash.
* Try a lower --baud rate if you get any errors. Default is 115200.

To write back:

```
esptool.py --baud 920600 write_flash --flash_mode dio --flash_size 4MB 0 original_contents.bin
```

> So that we can program, read back in file and do a binary compare of all the files...

Sure, you can do this using the above steps. Also, the write_flash step used by esp-idf `make flash` also verifies the MD5 hash of all data written out, so it's very unlikely that write_flash will succeed but the data will be incorrect. However it can't hurt to be sure!

JPB1JPB1, 2016-10-04:
Ok thanks, Mr projectgus

closed 2016-10-05 by projectgus

Zibri, 2020-05-29:
does anybody have a dump of an ESP32-wroom32 factory image?
not the AT one.. the one originally flashed in the device

commit 03c59658 (2018-11-28): Merge branch 'bugfix/test_idf_monitor_hangs' into 'master'

tools: Fix idf_monitor test case occasional hangup or failure

Closes #39

See merge request idf/esp-idf!3651