espressif/esp-idf #41: SDK tool configuration, example for the path for xtensa-esp32-elf- in menuconfig ?

JPB1JPB1, 2016-10-04:
Hello Guys,

Running under W10 and Msys2, following few guides lines, and running the menuconfig for the examples 01_ xx to 05_xxx, but when i try the make to compile any examples , it looks that the path for the "gcc" is not right.. or at CC log.o is hangs if running ...
IDF_PATH is ok. with the export  IDF_PATH="my_drive/my_directory/esp-idf" and echo $IDF_PATH display it.
Struggling around this , please can you precise the default path for xtensa-esp32-elf-gcc , 

It looks like this and hangs : 
make[2] : on entre dans le répertoire « /c/esp-idf/esp-idf/examples/01_hello_world/build/bootloader/log »
CC log.o
..
I've missed something ... Any idea ?
Have a great day.
Jp

projectgus, 2016-10-05:
Can you please post the full output of 'make V=1'? Thanks.

projectgus, 2016-10-05:
xtensa-esp32-elf-gcc should be on your path, you can check by running the command `which xtensa-esp32-elf-gcc`.

There are some issues reported using MSYS2 with certain Windows system services. If the build appears to hang and you open task manager, click on "Processes from All Users" and there is a svchost.exe using 100% CPU, then this may be the issue.

I saw this in esp-idf a few times (didn't hang totally, just slowed the build down) but I haven't fully investigated it yet. A very temporary workaround is to kill the svchost.exe task that's using 100% CPU. A better temporary workaround is to right-click, choose "Go to Service", and Stop the associated Service.

JPB1JPB1, 2016-10-05:
Hello

First, i've restarted a full installation of everything, we need about a minimum of 2GByte for all ...

the command  :  which xtensa-esp32-elf-gcc
$ which xtensa-esp32-elf-gcc
/opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc

Full copy of my "first" make of esp-idf-template using using the menuconfig ( i've not saved but it runs ok!)

----------------------------------------------------
$ make
GENCONFIG
*
* Restart config...
*
*
* ESP32-specific config
*
CPU frequency
  1. 80 MHz (ESP32_DEFAULT_CPU_FREQ_80)
  2. 160 MHz (ESP32_DEFAULT_CPU_FREQ_160)
> 3. 240 MHz (ESP32_DEFAULT_CPU_FREQ_240)
choice[1-3?]: 3
Select stack to enable (WiFi or BT)
> 1. WiFi (ESP32_ENABLE_STACK_WIFI)
  2. BT (ESP32_ENABLE_STACK_BT)
  3. None (ESP32_ENABLE_STACK_NONE)
choice[1-3?]: 1
Reserve memory for two cores (MEMMAP_SMP) [Y/n/?] y
Use TRAX tracing feature (MEMMAP_TRACEMEM) [N/y/?] n
Use external SPI SRAM chip as main memory (MEMMAP_SPISRAM) [N/y/?] n
System event queue size (SYSTEM_EVENT_QUEUE_SIZE) [32] 32
Event loop task stack size (SYSTEM_EVENT_TASK_STACK_SIZE) [2048] 2048
Main task stack size (MAIN_TASK_STACK_SIZE) [4096] 4096
Standard-out output adds carriage return before newline (NEWLIB_STDOUT_ADDCR) [Y                                                                                                                                                                                               /n/?] y
Enable Ultra Low Power (ULP) Coprocessor (ULP_COPROC_ENABLED) [N/y/?] (NEW) N
*
* FreeRTOS
*
Run FreeRTOS only on first core (FREERTOS_UNICORE) [Y/n/?] y
Xtensa timer to use as the FreeRTOS tick source
> 1. Timer 0 (int 6, level 1) (FREERTOS_CORETIMER_0)
  2. Timer 1 (int 15, level 3) (FREERTOS_CORETIMER_1)
  3. Timer 2 (int 16, level 5) (FREERTOS_CORETIMER_2)
choice[1-3?]: 1
Tick rate (Hz) (FREERTOS_HZ) [1000] 1000
Halt when an SMP-untested function is called (FREERTOS_ASSERT_ON_UNTESTED_FUNCTI                                                                                                                                                                                               ON) [Y/n/?] (NEW) Y
Check for stack overflow
  1. No checking (FREERTOS_CHECK_STACKOVERFLOW_NONE)
> 2. Check by stack pointer value (FREERTOS_CHECK_STACKOVERFLOW_PTRVAL)
  3. Check using canary bytes (FREERTOS_CHECK_STACKOVERFLOW_CANARY)
choice[1-3?]: 2
Amount of thread local storage pointers (FREERTOS_THREAD_LOCAL_STORAGE_POINTERS)                                                                                                                                                                                                [3] 3
Panic handler behaviour
  1. Print registers and halt (FREERTOS_PANIC_PRINT_HALT)
> 2. Print registers and reboot (FREERTOS_PANIC_PRINT_REBOOT)
  3. Silent reboot (FREERTOS_PANIC_SILENT_REBOOT)
  4. Invoke GDBStub (FREERTOS_PANIC_GDBSTUB)
choice[1-4?]: 2
Make exception and panic handlers JTAG/OCD aware (FREERTOS_DEBUG_OCDAWARE) [Y/n/                                                                                                                                                                                               ?] y
FreeRTOS assertions
> 1. abort() on failed assertions (FREERTOS_ASSERT_FAIL_ABORT)
  2. Print and continue failed assertions (FREERTOS_ASSERT_FAIL_PRINT_CONTINUE)
  3. Disable FreeRTOS assertions (FREERTOS_ASSERT_DISABLE)
choice[1-3?]: 1
Stop program on scheduler start when JTAG/OCD is detected (FREERTOS_BREAK_ON_SCH                                                                                                                                                                                               EDULER_START_JTAG) [Y/n/?] y
#
# configuration written to /c/esp-idf-template/esp-idf-template/sdkconfig
#
make[1] : on entre dans le répertoire « /c/esp-idf/esp-idf/components/bootloader                                                                                                                                                                                               /src »
GENCONFIG
#
# configuration written to /c/esp-idf/esp-idf/components/bootloader/src/sdkconfi                                                                                                                                                                                               g
#
make[2] : on entre dans le répertoire « /c/esp-idf-template/esp-idf-template/bui                                                                                                                                                                                               ld/bootloader/log »
CC log.o

------------------------------------------------------

===> here it hangs for a while and no problem in svchost ....... No processus, service are at 100% ...  .....

Hope to solve this also.
Jp

JPB1JPB1, 2016-10-05:
Hello all

it is solved.

By excluding 3 directory ( msys32 and the 2 esp-idf-xxx repertory)  from the Avast antivirus, the "make" command runs.    

I've got a trace of the first compile if needed to show for the other..

I get 3 *.bin files in their build directory... 

Good news. and best regards
Jp

closed 2016-10-05 by JPB1JPB1