espressif/esp-idf #44: printf stops printing after first newline

jmattsson, 2016-10-06:
It appears that for some strange reason the regular C `printf()` function stops printing when it encounters the first newline (`\n`) in a string. This can be trivially reproduced by attempting to print two lines in the one call; see below for a patch to the official 01_hello_world example.

```
diff --git i/examples/01_hello_world/main/hello_world_main.c w/examples/01_hello_world/main/hello_world_main.c
index ad2c0ac..ca907ea 100644
--- i/examples/01_hello_world/main/hello_world_main.c
+++ w/examples/01_hello_world/main/hello_world_main.c
@@ -14,7 +14,7 @@
 
 void hello_task(void *pvParameter)
 {
-    printf("Hello world!\n");
+    printf("Hello world!\nHiding in plain sight...\n");
     for (int i = 10; i >= 0; i--) {
         printf("Restarting in %d seconds...\n", i);
         vTaskDelay(1000 / portTICK_RATE_MS);
```
Despite applying the above patch, the output stays at:
```
Hello world!
```

If using the `%s` modifier, a newline in the argument string also causes `printf()` to stop outputting characters, effectively cutting short the output. E.g.
```
printf("Hello%s world!\n", "Hiding\n in plain sight...");
```
yields the following output:
```
HelloHiding
```

projectgus, 2016-10-06:
That @jmattsson . I committed a patch for this into our review queue yesterday, it's a bug in the write syscall.

Because everyone's on holidays this week I'm not sure if we'll get the merged to github until next week (review policy), but I might be able to pop up a temporary branch.

jmattsson, 2016-10-06:
Really? I instrumented `_write_r()` but by the time it got there the string had already been truncated at the `\n`.

projectgus, 2016-10-06:
Yeah it's not an obvious bug though., _write_r should return number of bytes written, but doesn't. Stdout is line buffered by default: so the first line gets sent to _write_r & TXed,, _write_r returns zero, newlib discards the rest of the buffer thinking there was an error.

Fix is incoming now, should be auto-pushed github shortly...

closed 2016-10-06 by igrr

jmattsson, 2016-10-06:
Ah, sneaky bug! Glad it was an easy fix though.

commit 12caaed2 (2016-10-06): Merge branch 'bugfix/syscall_write' into 'master'

syscall write: Should return number of bytes written

Fixes bug where sometimes output truncates after a newline, or large
chunks of large output buffers are lost.

Closes #44

See merge request !133
commit 21b39193 (2018-11-29): Merge branch 'bugfix/vfs_open_write_close_perf_test' into 'master'

Fix VFS unit test failure ("Open & write & close through VFS passes performance test")

Closes #44

See merge request idf/esp-idf!3759