espressif/esp-idf #45: Suggestion: Rename environment variable IDF_PATH

nkolban, 2016-10-06:
The ESP-IDF framework is using the environment variable `IDF_PATH`.  I can't help but feel that this is too "generic".  I was watching a colleague look at some of my ESP files today and many of my variables were `ESP_XXX` except `IDF_PATH`.  If I were to make a suggestion, it would be `ESP_IDF_PATH`.  It just feels to me that our ESP32 core prefix is "ESP" and if that were the case, `ESP_IDF_PATH` would fit in that story just fine.

bvernoux, 2016-10-06:
Yes I'm totally agree ESP_IDF_PATH will be more adapted

projectgus, 2021-01-11:
Closing this issue. I agree this argument makes some sense, but we're pretty standardized on `IDF_` prefix and especially `IDF_PATH` is now very widely used - so we're unlikely to change it.

closed 2021-01-11 by projectgus