espressif/esp-idf #47: Build Toolchain

vbextreme, 2016-10-07:
I'm try to build toolchain for esp32 on Odroid U3.
I am following the guide below [https://github.com/espressif/esp-idf/blob/master/docs/linux-setup.rst](url) but GMP return error
```
[INFO ]  Installing GMP for host
[ERROR]    make[3]: *** [div_qr_1n_pi1.lo] Error 1
[ERROR]    make[2]: *** [all-recursive] Error 1
[ERROR]    make[1]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing GMP for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gmp_backend[scripts/build/companion_libs/100-gmp.sh@106]
[ERROR]  >>        called from: do_gmp_for_host[scripts/build/companion_libs/100-gmp.sh@62]
[ERROR]  >>        called from: do_companion_libs_for_host[scripts/build/companion_libs.sh@36]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-59-g8d95cad/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 4:08.63)
[04:10] / make: *** [build] Errore 2
```
I have already installed in my machine and gmp I also compiled the 6.1 without ever having problems, I think there is some configuration problem.
Some advice?

nkolban, 2016-10-07:
Is there a more detailed build log that can be shared?

vbextreme, 2016-10-07:
I have copy gmp 6.1 into 6.0 and now building, also it works with 5.1.3
```
./ct-ng xtensa-esp32-elf
./ct-ng menuconfig
```
select "companion libraries" -> GMP version -> 5.1.3
save ".config"
```
./ct-ng build
```

I expect to finish the build to confirm that everything works

vbextreme, 2016-10-08:
I made several attempts to obtain the following result:
```
GCC 4.8       + GMP 5.1.3 = Build OK
GCC 5.2       + GMP 6.1.0 = (fake install gmp 6.1) Error on Installing pass-2 core C gcc compiler
GCC 5.2/4.8 + GMP 6.0.0 = Error on Installing GMP for host
GCC 5.3       + GMP 6.1.0 = (fake install gcc 5.3 and gmp 6.1) Error patch gcc :(
```

I'm trying again to compile gcc 5.2 and gmp 6.1.0 but I do not think anything will change.
Later I will compile with gcc 5.2 gmp 5.1.3

[gcc5.2_gmp6.0_build.txt](https://github.com/espressif/esp-idf/files/517709/gcc5.2_gmp6.0_build.txt)
[gcc5.2_gmp6.1_build.txt](https://github.com/espressif/esp-idf/files/517710/gcc5.2_gmp6.1_build.txt)
[gcc5.3_gmp6.1.0_build.txt](https://github.com/espressif/esp-idf/files/517711/gcc5.3_gmp6.1.0_build.txt)

quoclap, 2016-10-18:
Try to use this command : 
> sed -i.bak '/__need_size_t/d' .build/src/gmp-5.1.3/gmp-h.in

then use this comand again:

> ./ct-ng build

projectgus, 2016-10-20:
At the moment gcc 4.8 is the only supported toolchain. We're working on gcc 5 support but you should expect build problems like the ones you are seeing (and you may encounter code generation problems as well.) I recommend using the gcc 4.8 you've built, at least for now.

projectgus, 2016-12-22:
I'm going to close this, but please reopen if there are still outstanding problems with supported toolchain configurations.

commit 2c7fc07a (2018-11-29): Merge branch 'bugfix/various_ldgen_fixes' into 'master'

Fix multiple ldgen issues

Closes #47

See merge request idf/esp-idf!3799
closed 2016-12-22 by projectgus