espressif/esp-idf #50: Downloading the program when MD5 parity error

eboxmaker, 2016-10-10:
A few days ago can also be a normal download. This download error occurs after updating the firmware library
`Flashing project app to 0x10000...
esptool.py v2.0-dev
Connecting...
Uploading stub...
Running stub...
Stub running...
Attaching SPI flash...
Configuring flash size...
Erasing flash...
Unc size 4224 comp size 2635 comp blocks 3
Wrote 2635 bytes at 0x00001000 in 0.2 seconds (90.3 kbit/s)...
File  md5: b8a3654c3478dbe4ee99e418fac4c5b5
Flash md5: b846cfdf358a11fdf47a71901ec3cda1

A fatal error occurred: MD5 of file does not match data in flash!
make: *** [C:/esp32_idf/esp-idf/components/esptool_py/Makefile.projbuild:28：flash] 错误 2`

projectgus, 2016-10-10:
Hi @eboxmaker,

Thanks for reporting this. Do you consistently get the same "Flash md5" value when the flashing fails?

Would it be possible you to please attach the `build/bootloader/bootloader.bin` file to this issue, if it doesn't contain any sensitive modifications?

Angus

eboxmaker, 2016-10-10:
The MD5 errors are the same for every download.
I deleted the` build` folder. Recompile the program to download. The same error occurs.
 I did not make any changes to the program.

projectgus, 2016-10-10:
OK, thanks clarifying. Can you attach the bootloader.bin file here, please? Mine has a different md5sum, I'm wondering if the bug has something to do with the content.

widora, 2016-10-10:
eboxmaker upload by me [bootloader.zip](https://github.com/espressif/esp-idf/files/518723/bootloader.zip)

projectgus, 2016-10-11:
Hi folks,

I've tried flashing this bootloader image on Windows and Linux, including under current master and esp-idf V0.9. No problems. So I have some follow-up questions to try and narrow it down:

* What serial port are you using? If you run `make flash V=1` can you post the full esptool.py command line here, please?
* What ESP32 hardware do you have connected?
* If you disable "Compressed Upload" in `make menuconfig` serial flasher settings, does anything change?
* If you lower the baud rate (for a very "safe" low value try 9600) in `make menuconfig`, does anything change?

Also, so you know, someone has posted on the [esp32.com forum with the same error and md5 values](http://esp32.com/viewtopic.php?f=2&t=322). I'm guessing this is someone else with the same bug.

Thanks,

Angus

projectgus, 2016-10-13:
Rather than answer the above questions, can you please esptool.py from this link and report back if flashing succeeds? Thanks.
https://github.com/themadinventor/esptool/tree/bugfix/write_fails_cmp_bit_set

(There are several ways to do this: You can either download esptool.py and run it in place, or you can copy it into the esp-idf/components/esptool_py/esptool directory, or you can navigate to this directory and use git to checkout that branch into the submodule.)

AnalogLamb, 2016-10-13:
@projectgus 
[image]

AnalogLamb, 2016-10-13:
the error is same.

Unc size 4224 comp size 2635 comp blocks 3
Wrote 2635 bytes at 0x00001000 in 0.2 seconds (90.3 kbit/s)...

[image]

projectgus, 2016-10-14:
Hi folks,

Thanks for updating me with the unchanged output. Can you please grab the newest update on the same branch:
https://github.com/themadinventor/esptool/tree/bugfix/write_fails_cmp_bit_set

And try running these two commands:

```
esptool.py --port PORT read_flash_status
esptool.py --port PORT flash_id
```

(Please copy or screenshot the output and post it here.)

If that all seems to go well, you can try:

```
esptool.py --port PORT write_flash_status --non-volatile 0
```

... and see if you can flash again. Even if it works, could you please post the output from the first two commands - they will be very helpful for tracking down the underlying problem.

Thanks for your patience.

(PS You can reply here or the Github Issue, it's not necessary to reply in both places - I'm watching both.)

AnalogLamb, 2016-10-14:
I just copy the esptool.py to the directory esp-idf/components/esptool_py/esptool, and replace the old esptool.py, It's successful

[image]

AnalogLamb, 2016-10-14:
OK, the result of esptool.py command as blelow.

[image]

projectgus, 2016-10-25:
Thanks for all the debug output and testing assistance. This bug should be fixed since commit f988dc9 as linked above.

closed 2016-10-25 by projectgus

architmuchhal12, 2017-05-26:
Hey,

I am trying to access the following link to get the new _esptool.py_ because I am encountering the MD5 Flashing error. Unfortunately, the link is not working and showing me "Page not found".

[](https://github.com/themadinventor/esptool/tree/bugfix/write_fails_cmp_bit_set)

Can anyone post the working link?
Thanks!

greeshmarepaka, 2018-02-02:
How to getrid of the error "MD5 of file does not match data in flash!"

joicetm, 2018-02-13:
Hi,

i am also getting the same error with my custom esp32 pico board. i have tried different idf versions ,2.1, 3.0 .
would it be a hardware issue?

ctnalves, 2018-03-08:
Hi, 
I had the same problem with a esp32. Than I just disconect all the cables conected to the I/O pins and I tryied to upload again. It worked!

veestrc, 2018-04-20:
Disconnecting all all wires connected to pins worked for me also...
Thanks https://github.com/ctnalves

BenUniqcode, 2018-06-20:
I also ran into this problem and thanks to this thread found that removing the IO connections solved it. But I only had GPIO 12 and 13 connected. I don't see why these would interfere with uploading?

igrr, 2018-06-20:
GPIO12 is a bootstrapping pin which sets flash voltage.

BenUniqcode, 2018-06-20:
@igrr Ok thanks. How are we meant to know that? The pinout diagram (http://www.heltec.cn/download/WIFI_Kit_32-Diagram.pdf) doesn't mention it. 
I really think it should clearer which "GPIO" pins are not actually usable as GPIO pins!

igrr, 2018-06-20:
This is mentioned, for example, in ESP32 datasheet, section 2.4 "Strapping pins". Note at the end of the section says that GPIO12 needs to have low logic level at startup. Alternatively, if you can not guarantee this, you can permanently set flash voltage to 3.3V, by burning efuses using espefuse.py tool. See https://github.com/espressif/esptool/wiki/espefuse#setting-flash-voltage-vdd_sdio for details.
Once EFUSE is written, you don't need to ensure any specific level of GPIO12 on reset.

projectgus, 2018-06-20:
@BenUniqcode Please consider forwarding this feedback to Heltec as well, so they can update the diagram for their board to include strapping pins.

dave88PL, 2020-10-30:
Hi,
PIN 12 was a problem in my case. On one device it worked fine, on second one - I hade this error.