espressif/esp-idf #52: printf support for 64-bit integers

rojer, 2016-10-14:
According to [reports from our users](https://github.com/cesanta/mongoose/issues/717), ESP32 continues to have the same problem with its build of Newlib as ESP8266 used to have: there is no support for printing floats or 64-bit ints.
Would it be possible to enable this support? It only takes [3 lines](https://github.com/cesanta/esp-open-sdk/commit/c18ea05dfb90d4a15c6517a1bfeda3892f7e25dd) to enable the extra formats, but it makes life so much easier.
FWIW, some other vendors (e.g. TI) ship Newlib with this support on, and the manifestations of this bug are very nasty - no compiler errors or warnings, printf just prints garbage at runtime.

igrr, 2016-10-14:
Floating point printf should work in esp-idf (i.e. we have a test for it which is currently passing; if you see an issue with floats please open a ticket and provide some code to reproduce).

Long long support was not enabled when building newlib however, so changing this setting would require replacing considerable part of newlib functions in ROM with modified copies in flash/IRAM. We can probably offer that as a menuconfig option.

projectgus, 2016-10-14:
To bring us full circle with this discussion, the first part was @nkolban's post on the ESP32 forum here:
http://esp32.com/viewtopic.php?f=13&t=333

> It only takes 3 lines to enable the extra formats, but it makes life so much easier.

Our newlib was [configured with --enable-newlib-nano-formatted-io](https://gist.github.com/igrr/92f6115dd0986b3a82a46d2ba729b519) - which pulls in [nano-vfprintf_i.c](https://github.com/espressif/newlib-esp32/blob/master/newlib/libc/stdio/nano-vfprintf_i.c). As far as I know, the only way to support 64-bit formatting is to disable "nano" and go back entirely to the default I/O formatting implementation.

As Ivan says though, this is probably something that could be enabled via a menuconfig option.

rojer, 2016-10-14:
for a time we tried to work around it, but it kept popping up in different places. whatever modest savings in terms of code size it brings, it's not worth the time wasted debugging the subtle issues it causes. there may have been an argument for size on ESP8266, but on 32 it should not matter as much.
please, just give the users working printf :)

igrr, 2016-10-14:
In retrospect, I should have stubbed `_vprintf_r` family of functions through a pointer table, and put both implementations (nano and full) into ROM.

Will add full formatting implementation (selectable via menuconfig) as a work-around.

rojer, 2016-10-14:
(sorry, just noticed first response by @igrr)
ok, the ROM argument is a bit different. though i'd still say - just do it. the device is tiny, but it can still manipulate 64-bit numbers - various IDs, large counters... we've ran into this in many places.
at the very least make compiler warn about long long formats. really, debugging these issues is no fun.

rojer, 2016-10-14:
ok, thank you! making that option default to on, for those who know what they're doing and want to save extra few bytes to turn it off would be ideal.

negativekelvin, 2016-10-14:
Curious to know what the IRAM penalty will amount to

rojer, 2016-10-14:
don't have ESP32 env set up to test, but on ESP8266 newib's "fat" `_vfprintf_r` is ~6.5K

rojer, 2016-12-02:
any news on this? i'm starting work on our esp32 port and i'm interested now :)

igrr, 2016-12-02:
I haven't started working on this yet.

rojer, 2016-12-02:
ok. in case i wanted to do it myself... i see that https://github.com/espressif/esp-idf/tree/master/components/newlib only includes binary blobs. you don't provide sources for newlib used in IDF?

igrr, 2016-12-02:
Newlib is here: https://github.com/espressif/newlib-esp32

igrr, 2016-12-04:
@rojer FYI, I did the necessary changes to support 64-bit formats, it should be reviewed and merged this week.

rojer, 2016-12-04:
great, thank you!

rojer, 2016-12-13:
@igrr, gentle ping. if it's really close, can you give it a little push? thanks!

igrr, 2016-12-13:
It's been merged. But we broke master with another change (interrupt allocation), hence you don't see any update for 4 days. Once we fix this (hopefully today), it's going to be deployed automatically. Sorry for the delay!

rojer, 2016-12-13:
perfect timing :) thanks

closed 2016-12-13 by igrr

ygoe, 2022-03-19:
Okay, so how does it work now?

igrr, 2022-03-20:
@ygoe It does work with C standard library functions (such as `printf` and `fprintf`) by default.

If you are seeing an issue, please make sure [CONFIG_NEWLIB_NANO_FORMAT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-newlib-nano-format) option isn't enabled in you sdkconfig.

Note that 64-bit formatting is not supported in `ets_printf` or `esp_rom_printf` functions, as well as `ESP_EARLY_LOGx` macros.

ygoe, 2022-03-20:
Hm. I found it out myself meanwhile. I was looking for this format string placeholder: It's not mentioned anywhere around and there are several unverified or broken suggestions on the web.

    ESP_LOGI(TAG, "Number: %lld", i64value);