espressif/esp-idf #53: Both esp-idf-template and 03_http_request crashes constantly

imxieyi, 2016-10-14:
I compiled the code with the latest esp-idf and esp-wifi-lib under msys2. But when I run it on esp32, it constantly crashes. If I remove codes about wifi connection, it won't crash.
The logs are below:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:920
load:0x40078000,len:2724
ho 0 tail 12 room 4
load:0x40098000,len:508
entry 0x40098118
[0;32mI (94) heap_alloc_caps: Initializing heap allocator:[0m
[0;32mI (95) heap_alloc_caps: Region 19: 3FFC0804 len 0001F7FC tag 0[0m
[0;32mI (96) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1[0m
[0;32mI (106) cpu_start: Pro cpu up.[0m
[0;32mI (111) cpu_start: Starting app cpu, entry point is 0x400809d4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (127) cpu_start: Pro cpu start user code[0m
rtc v112 Sep 26 2016 22:32:10
XTAL 40M
[0;32mI (163) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (43) cpu_start: Starting scheduler on APP CPU.[0m
frc2_timer_task_hdl:3ffc5c48, prio:22, stack:2048
tcpip_task_hdlxxx : 3ffc6940, prio:18,stack:2048
Guru Meditation Error of type IllegalInstruction occurred on core   0. Exception was unhandled.
Register dump:
PC      :  400856cb  PS      :  00060530  A0      :  80087795  A1      :  3ffc2450  
A2      :  0abd1faa  A3      :  3ffbe078  A4      :  3ffc2470  A5      :  000000fc  
A6      :  00000003  A7      :  3ffae0c0  A8      :  3f400094  A9      :  3ffc2430  
A10     :  00000000  A11     :  00000001  A12     :  00000000  A13     :  0010ef2a  
A14     :  00000006  A15     :  00000000  SAR     :  0000001a  EXCCAUSE:  00000000  
EXCVADDR:  00000000  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
Rebooting...

If I use a higher log level, the error type will be unreadable.
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:920
load:0x40078000,len:2724
ho 0 tail 12 room 4
load:0x40098000,len:508
entry 0x40098118
[0;32mI (94) heap_alloc_caps: Initializing heap allocator:[0m
[0;32mI (94) heap_alloc_caps: Region 19: 3FFC0804 len 0001F7FC tag 0[0m
[0;32mI (95) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1[0m
[0;32mI (105) cpu_start: Pro cpu up.[0m
[0;32mI (111) cpu_start: Starting app cpu, entry point is 0x400809d4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (126) cpu_start: Pro cpu start user code[0m
rtc v112 Sep 26 2016 22:32:10
XTAL 40M
[0;32mI (162) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (43) cpu_start: Starting scheduler on APP CPU.[0m
D (43) nvs: init start=6 count=3[0m
D (2033) nvs: nvs_open misc 1[0m
D (2033) nvs: nvs_get_str_or_blob log[0m
frc2_timer_task_hdl:3ffc5ddc, prio:22, stack:2048
tcpip_task_hdlxxx : 3ffc6940, prio:18,stack:2048
phy_version: 123, Sep 13 2016, 20:01:58, 0
D (2213) nvs: nvs_open nvs.net80211 1[0m
D (2213) nvs: nvs_get opmode 1[0m
D (2243) nvs: nvs_get country 1[0m
D (2263) nvs: nvs_get_str_or_blob sta.ssid[0m
D (2313) nvs: nvs_get_str_or_blob sta.mac[0m
D (2353) nvs: nvs_get sta.authmode 1[0m
D (2373) nvs: nvs_get_str_or_blob sta.pswd[0m
D (2433) nvs: nvs_get_str_or_blob sta.pmk[0m
D (2473) nvs: nvs_get sta.chan 1[0m
D (2493) nvs: nvs_get auto.conn 1[0m
D (2513) nvs: nvs_get bssid.set 1[0m
D (2533) nvs: nvs_get_str_or_blob sta.bssid[0m
D (2573) nvs: nvs_get sta.phym 1[0m
D (2593) nvs: nvs_get sta.phybw 1[0m
D (2613) nvs: nvs_get_str_or_blob sta.apsw[0m
D (2653) nvs: nvs_get_str_or_blob sta.apinfo[0m
D (2903) nvs: nvs_get_str_or_blob ap.ssid[0m
D (2953) nvs: nvs_get_str_or_blob ap.mac[0m
D (2993) nvs: nvs_get_str_or_blob ap.passwd[0m
D (3053) nvs: nvs_get_str_or_blob ap.pmk[0m
D (3103) nvs: nvs_get ap.chan 1[0m
D (3123) nvs: nvs_get ap.authmode 1[0m
D (3143) nvs: nvs_get ap.hidden 1[0m
D (3163) nvs: nvs_get ap.max.conn 1[0m
D (3183) nvs: nvs_get bcn.interval 2[0m
D (3203) nvs: nvs_get ap.phym 1[0m
D (3223) nvs: nvs_get ap.phybw 1[0m
D (3243) nvs: nvs_get ap.sndchan 1[0m
D (3263) nvs: nvs_set_blob sta.mac 6[0m
D (3373) nvs: nvs_set_blob ap.mac 6[0m
pp_task_hdl : 3ffca19c, prio:23, stack:8192
[0;32mI (3453) example: Setting WiFi configuration SSID ForIOT...[0m
mode : sta(24:0a:c4:01:61:dc)
D (3453) event: SYSTEM_EVENT_STA_START[0m
n:6 2, o:1 0, ap:255 255, sta:6 2, prof:1
state: 0 -> 2 (b0)
Guru Medi}atioj Ekrcr cf dype IllegmlAnctructaoo occurred on core   0. Exceptiob gas un``fdhad.
Register du``*
@C   : 400df666  PC   : 00060730  A0      : 800dfff4  A1      : 3ffca000  
A2      : 00000000  A3      : 3ffbcd54  A5      : 3ffb6808  
A6      : 3ffca720  A8      : 00000010  A9      : 3ffca000  
A10     : 00000060  A11     : 00000000  A12     : 00000000  A13     : 000000b0  
A14     : 3ffb0e68  A15     : 00000080  SAR     : 00000017  EXCCAUSE: 00000000  
EXCVADDR: 00000000  LBEG    : 4000c2e0  LCOUNT  : ffffffff  
Rebgoting...

Spritetm, 2016-10-14:
What hardware, and specifically what power supply are you using? This smells very much like a power supply issue, to be honest.

imxieyi, 2016-10-14:
I use ESP-32S from Ai-thinker. And I use a CH340G usb-ttl as power supply.

Spritetm, 2016-10-14:
Can you see what happens if you use a more powerful 3.3V power supply? Try to get something that can supply at least 500mA, to be sure.

imxieyi, 2016-10-14:
I switched to a 1A usb charger and it now works! Thank you!
But I wonder if SoftAP mode consumes less energy than Station mode because the initial firmware of this board, which includes a softAP runs smoothly with the power supply of CH340G.

Spritetm, 2016-10-14:
It's hard to say, because your ESP32 dies because of a spike of power usage. It could be that the average power usage is much lower, but associating the ESP32 with an AP may cause a single spike in power use causing the supply voltage to drop and the entire thing to become unstable.

Anyway, glad you could make it work!.

closed 2016-10-14 by Spritetm