espressif/esp-idf #57: Build idf example MacOSX

quoclap, 2016-10-18:
Hi,
I'm trying to build example idf example projects. I've setup my mac using guide : 
https://github.com/espressif/esp-idf/blob/master/docs/macos-setup.rst
But when go to menuconfig using 

> make menuconfig

 I met the error, is there any missing dependencies i've missed?

> Trans-MacBook-Pro:~ quoclap$ cd esp/esp-idf/examples/02_blink/
Trans-MacBook-Pro:02_blink quoclap$ export IDF_PATH=~/esp/esp-idf
Trans-MacBook-Pro:02_blink quoclap$ make menuconfig
MAKEFLAGS="" \
	CC=cc LD=ld \
	/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/quoclap/esp/esp-idf/tools/kconfig
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE -lintl 
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lintl 
Undefined symbols for architecture x86_64:
  "_acs_map", referenced from:
      _dialog_checklist in checklist.o
      _print_arrows in checklist.o
      _dialog_clear in util.o
      _draw_box in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      ...
  "_cbreak", referenced from:
      _init_dialog in util.o
  "_delwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_doupdate", referenced from:
      _dialog_checklist in checklist.o
  "_endwin", referenced from:
      _init_dialog in util.o
      _end_dialog in util.o
  "_flash", referenced from:
      _dialog_inputbox in inputbox.o
  "_getcurx", referenced from:
      _init_dialog in util.o
      _print_autowrap in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_arrows in menubox.o
  "_getcury", referenced from:
      _init_dialog in util.o
      _print_autowrap in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_arrows in menubox.o
  "_getmaxx", referenced from:
      _show_help in mconf.o
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      ...
  "_getmaxy", referenced from:
      _dialog_checklist in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      ...
  "_has_colors", referenced from:
      _color_setup in util.o
      _draw_shadow in util.o
  "_init_pair", referenced from:
      _init_one_color in util.o
  "_initscr", referenced from:
      _init_dialog in util.o
  "_keypad", referenced from:
      _dialog_checklist in checklist.o
      _init_dialog in util.o
      _on_key_esc in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
      ...
  "_newwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_nodelay", referenced from:
      _on_key_esc in util.o
  "_noecho", referenced from:
      _init_dialog in util.o
  "_scrollok", referenced from:
      _dialog_checklist in checklist.o
      _dialog_menu in menubox.o
      _do_scroll in menubox.o
  "_start_color", referenced from:
      _color_setup in util.o
  "_stdscr", referenced from:
      _show_help in mconf.o
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _init_dialog in util.o
      _end_dialog in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      ...
  "_subwin", referenced from:
      _dialog_checklist in checklist.o
      _dialog_textbox in textbox.o
      _dialog_menu in menubox.o
  "_ungetch", referenced from:
      _on_key_esc in util.o
  "_waddch", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_button in util.o
      ...
  "_waddnstr", referenced from:
      _print_item in checklist.o
      _print_arrows in checklist.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_autowrap in util.o
      _print_button in util.o
      _dialog_inputbox in inputbox.o
      ...
  "_wattrset", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _print_title in util.o
      _print_button in util.o
      ...
  "_wbkgdset", referenced from:
      _dialog_textbox in textbox.o
      _print_position in textbox.o
      _dialog_menu in menubox.o
  "_wclrtoeol", referenced from:
      _print_line in textbox.o
      _do_print_item in menubox.o
  "_wgetch", referenced from:
      _dialog_checklist in checklist.o
      _on_key_esc in util.o
      _dialog_inputbox in inputbox.o
      _dialog_textbox in textbox.o
      _dialog_yesno in yesno.o
      _dialog_menu in menubox.o
  "_winch", referenced from:
      _draw_shadow in util.o
  "_wmove", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_arrows in checklist.o
      _print_buttons in checklist.o
      _attr_clear in util.o
      _dialog_clear in util.o
      _end_dialog in util.o
      ...
  "_wnoutrefresh", referenced from:
      _dialog_checklist in checklist.o
      _dialog_clear in util.o
      _draw_shadow in util.o
      _dialog_textbox in textbox.o
      _print_page in textbox.o
      _dialog_menu in menubox.o
  "_wprintw", referenced from:
      _print_item in checklist.o
      _print_position in textbox.o
  "_wrefresh", referenced from:
      _dialog_checklist in checklist.o
      _print_item in checklist.o
      _print_buttons in checklist.o
      _end_dialog in util.o
      _dialog_inputbox in inputbox.o
      _print_buttons in inputbox.o
      _refresh_text_box in textbox.o
      ...
  "_wscrl", referenced from:
      _dialog_checklist in checklist.o
      _do_scroll in menubox.o
  "_wtouchln", referenced from:
      _attr_clear in util.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [mconf] Error 1
make: *** [/Users/quoclap/esp/esp-idf/tools/kconfig/mconf] Error 2

closed 2016-10-18 by quoclap

quoclap, 2016-10-18:
I've changed the tools/kconfig/lxdialog/check-lxdialog.sh file as andeepmistry's commit 
https://github.com/espressif/esp-idf/pull/42/commits/d087283fe168a32c0673f856bcf73ad420acaa37 
and the errors gone!
Thanks all!

endeebee, 2016-11-03:
I had the same issue, but after applying the suggested fix to check-lxdialog.sh, I still get errors:
`MAKEFLAGS="" \
	CC=cc LD=ld \
	/Library/Developer/CommandLineTools/usr/bin/make -C /Users/ndb/esp-idf-v0.9/tools/kconfig
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE -lncurses
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncurses
Undefined symbols for architecture x86_64:
  "_libintl_bindtextdomain", referenced from:
      _main in mconf.o
  "_libintl_gettext", referenced from:
      _conf in mconf.o
      _handle_exit in mconf.o
      _build_conf in mconf.o
      _conf_choice in mconf.o
      _conf_string in mconf.o
      _show_help in mconf.o
      _conf_save in mconf.o
      ...
  "_libintl_setlocale", referenced from:
      _main in mconf.o
  "_libintl_textdomain", referenced from:
      _main in mconf.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [mconf] Error 1
`

igrr, 2016-11-03:
@endeebee be sure to do `make clean` after applying the fix. Then do `make menuconfig` again.

endeebee, 2016-11-03:
I tried that, however the error already appears during make menuconfig

igrr, 2016-11-03:
The patch doesn't seem to be applied properly. There is no `-DKBUILD_NO_NLS` argument passed to cc on the command line. Please double check that you have added these lines:
https://github.com/espressif/esp-idf/pull/42/commits/d087283fe168a32c0673f856bcf73ad420acaa37#diff-1c6ecb5835068bd596b18b14d3b98df6R44

endeebee, 2016-11-03:
@igrr  Wow, thank you very much for your quick reply! I forgot to apply the second edit. it's working 👍