espressif/esp-idf #58: vTaskDelay / portTICK_RATE_MS wrong delay

bvernoux, 2016-10-19:
I have tested vTaskDelay(10 / portTICK_RATE_MS) using the sample (https://github.com/espressif/esp-idf/blob/master/examples/02_blink/main/blink.c) and the result is we obtain 15ms on the IO5 output instead of 10ms expected.
See Sigrok Signal Trace (done @40MHz sampling rate)
http://hydrabus.com/esp32_boot/esp32_bootflash_dump_dualspiflash_blinkled_10ms_sample_02_blink.sr

See screenshot here:
[image]

nkolban, 2016-10-19:
In your project, if you look inside sdkconfig what do you have for the values of:

- CONFIG_FREERTOS_HZ
- CONFIG_ESP32_DEFAULT_CPU_FREQ_240
- CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ

(Note: that I am not asking because I have any kind of an answer but rather I'm exercising the limited knowledge I have about speeds and timings).

bvernoux, 2016-10-19:
For information the full project is here
http://hydrabus.com/esp32_boot/esp32_bootflash_dump_dualspiflash_blinkled_10ms_sample_02_blink.7z
All is by default except I have added more info on debug output
CONFIG_FREERTOS_HZ=100
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240

bvernoux, 2016-10-19:
I have done more tests (changing CONFIG_FREERTOS_HZ to 1000) and we have a clean 10ms ON / 10 ms OFF signal on IO5 with a Jitter measured with oscilloscope(1GSPS mode with persistence ON) of +/- 100ns
I have also measured the Rise Time / Fall Time on IO5 GPIO and we have between 3.8ns & 4.4ns

I have done again the same test with CONFIG_FREERTOS_HZ=100 (I have added a printf() to display portTICK_RATE_MS and it display portTICK_RATE_MS=10) and I also measure with oscilloscope 10ms so in conclusion I think it was a bug during capture on sigrok / PulseView (or saleae logic analyzer...)
I have done also some other measurements with logic analyzer Saleae Logic 1.2.10 and I have not reproduced the issue which display 15ms instead of 10ms, anyway in future I will always check with oscilloscope to be sure ...

closed 2016-10-19 by bvernoux