espressif/esp-idf #59: Hang on post-reset boot?

hexagon5un, 2016-10-24:
I first noticed this behavior on [Arduino](https://github.com/espressif/arduino-esp32/issues/23), but it also occurs with the template app.  I almost don't believe this, but I've reproduced it too many times now.

Coming out of flashing the chip, when the reset button is pressed, the ESP32 doesn't seem to boot up right unless it's connected to a serial terminal. 

Steps to recreate:
1) Flash app-template (accepting all default options in menuconfig)
2) Press reset button
3) No blinking, no WiFi connection to network.
4) Run terminal program
5) Press reset
6) Connects to network, LED blinks.

Pressing reset many times without connecting a terminal program has no effect.  Powering the chip down and back up, however, makes it work regardless.

Just for fun, because I can't believe this issue, I attached a battery, flashed the firmware, pulled the USB cord, hit reset (nothing), re-attached the USB (nothing), and then opened up a serial terminal and hit reset.  Everything worked as it should.

It seems to be related to the up-from-reset flag and serial.  Though how the chip knows that I'm listening to it is beyond me.  Is there something in the configs that's halting on a test for a serial terminal?  How would it know?

Hardware is a WROOM dev board, esp-idf is all current as of today with a `git pull --recurse-submodules`.

igrr, 2016-10-24:
Some serial programs assert DTR and RTS when you open a serial port. DTR and RTS are wired via two transistors to an auto-reset circuit, which controls EN and GPIO0. My guess is that something is wrong with this circuit and EN/GPIO0 pins are in incorrect state for normal boot to happen. Can you measure voltage at EN and GPIO0 when no terminal is attached? Also, is the board a V1 or V2 board?

hexagon5un, 2016-10-24:
Hiya, sorry took so long.  

Board is dated 2016-08-31-30.

I think you hit it.  Both EN and GPIO 0 pull down to 0 V when a terminal application is running, but only to 1.415 V separately when it's not.  Pressing them both together pulls EN down to ~550 mV and GPIO 0 to 1.164 V.  

That should be enough to explain why it enters programming mode consistently, but only resets (EN low) when whatever DTR/RTS magic is going on.  I wasn't expecting analog problems!  :)

GPIO 0 also seems to float when EN is pulled low.  I accidentally touched it with my finger, and it'll wander around 1.2 - 1.6 V.  On releasing EN, GPIO 0 rises again, following EN by about 180 us.  (So there must be a delay in the test for that pin in the bootloader?)  FWIW.

Anyway, sounds like you've got a handle on the issue?  I almost always have a terminal open anyway, or boot up from power-down in applications, so I've never noticed it until now.  And now that it's clear what's going on, I'm all good.  Thanks.

ebutera, 2016-10-24:
I noticed the "inverse" too: run the app with a terminal open, close the terminal and the app will stop too.
wroom dev board with esp-idf v0.9

projectgus, 2016-12-22:
This sounds like it might be the blue Core Board v1, which had a problem with its auto-reset circuit that could lead to symptoms consistent to this.

If anyone is seeing the bug on Core Board v2 (black), please reopen.

closed 2016-12-22 by projectgus

charcoast, 2020-06-19:
Does this behavior have no software solution?

projectgus, 2020-06-24:
Hi @charcoast,

I think this issue was resolved by a development board hardware update back in 2016. Development boards made by Espressif since late 2016 do not have this issue. It's possible a third party development board has a similar issue, though.

Can you give some details about exactly what you're seeing?

dwonga, 2022-08-11:
Hi. I'm writing this in 2022. Just in case someone in the future have this weird problem in some cards such as mine (Lolin32_lite, esp32) and their program only works when a serial is connected. @igrr was right. The problem is when you use GPIO0 for some functions (in my case, I use GPIO0 for Touch1 touching input). The solution in this case was just not to use GPIO0 at all, and problem solved. I am sharing this in case it may be useful for someone.