espressif/esp-idf #60: make flash doesn't preserve serial port settings

nkolban, 2016-10-25:
When I execute `make flash` to flash an ESP32, I find that the serial port settings to my ESP32 aren't preserved after the flash completes.  Here is a listing of the settings before a flash:
```
$ stty -F /dev/ttyUSB0 -a
speed 115200 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^H; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z;
rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 100; time = 2;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk brkint ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon iexten -echo echoe echok -echonl -noflsh -xcase -tostop -echoprt echoctl echoke -flusho -extproc
```
and the same settings after executing a `make flash`:
```
$ stty -F /dev/ttyUSB0 -a
speed 921600 baud; rows 0; columns 0; line = 0;
intr = ^C; quit = ^\; erase = ^?; kill = ^H; eof = ^D; eol = <undef>; eol2 = <undef>; swtch = <undef>; start = ^Q; stop = ^S; susp = ^Z;
rprnt = ^R; werase = ^W; lnext = ^V; discard = ^O; min = 0; time = 0;
-parenb -parodd -cmspar cs8 -hupcl -cstopb cread clocal -crtscts
-ignbrk brkint ignpar -parmrk -inpck -istrip -inlcr -igncr -icrnl -ixon -ixoff -iuclc -ixany -imaxbel -iutf8
-opost -olcuc -ocrnl -onlcr -onocr -onlret -ofill -ofdel nl0 cr0 tab0 bs0 vt0 ff0
-isig -icanon -iexten -echo -echoe -echok -echonl -noflsh -xcase -tostop -echoprt -echoctl -echoke -flusho -extproc
```
I would have expected the settings to have been preserved.

negativekelvin, 2016-10-26:
Doesn't the port keep the settings from the last program that used/configured it?  Which would be esptool.py

nkolban, 2016-10-26:
I guess what I am then querying is the semantics that using esptool.py changes the existent settings of the serial port without restoring them to their original values at termination.

negativekelvin, 2016-10-26:
Seems like more times than not it would be preferred that the settings stay the same as when the port was last used.  Is there any precedent that a program stores and reverts the port settings?  It's not something I would expect.

projectgus, 2016-10-26:
It seems reasonable we should restore the original serial port settings on exit. This would probably be better opened as an issue against github.com/themadinventor/esptool, but I can track it from here.

projectgus, 2016-12-22:
Moved to an esptool feature request.

closed 2016-12-22 by projectgus