espressif/esp-idf #62: multiple definitions of uxPortCompareSet and xPortGetCoreID

nkolban, 2016-10-27:
I am compiling a mixed ESP-IDF / Arduino-ESP32 component application and am getting the following errors:

```
targets/esp32/jshardware.o: In function `uxPortCompareSet':
jshardware.c:(.text.uxPortCompareSet+0x0): multiple definition of `uxPortCompareSet'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.uxPortCompareSet+0x0): first defined here
targets/esp32/jshardware.o: In function `xPortGetCoreID':
jshardware.c:(.text.xPortGetCoreID+0x0): multiple definition of `xPortGetCoreID'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.xPortGetCoreID+0x0): first defined here
libs/network/esp32/jswrap_esp32_network.o: In function `uxPortCompareSet':
jswrap_esp32_network.c:(.text.uxPortCompareSet+0x0): multiple definition of `uxPortCompareSet'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.uxPortCompareSet+0x0): first defined here
libs/network/esp32/jswrap_esp32_network.o: In function `xPortGetCoreID':
jswrap_esp32_network.c:(.text.xPortGetCoreID+0x0): multiple definition of `xPortGetCoreID'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.xPortGetCoreID+0x0): first defined here
targets/esp32/main.o: In function `uxPortCompareSet':
main.c:(.text.uxPortCompareSet+0x0): multiple definition of `uxPortCompareSet'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.uxPortCompareSet+0x0): first defined here
targets/esp32/main.o: In function `xPortGetCoreID':
main.c:(.text.xPortGetCoreID+0x0): multiple definition of `xPortGetCoreID'
libs/network/esp32/network_esp32.o:network_esp32.c:(.text.xPortGetCoreID+0x0): first defined here
collect2: error: ld returned 1 exit status
Makefile:1995: recipe for target 'proj' failed
make: *** [proj] Error 1
```

When I look at the history, I see that the two functions causing issues are uxPortCompareSet and xPortGetCoreID.  It looks like their signatures changed recently.  I am not sure if this is a problem with ESP-IDF, Arduino-ESP32 or my attempt to mix the two.  I do notice that the Arduino-ESP32 seems to have an older "copy" of what appears to be compiled version of ESP-IDF with headers.

igrr, 2016-10-27:
Can you please post the last gcc command line before this error? (Run make V=1)

nkolban, 2016-10-27:
Ohh ... I just realized that this is a custom make ... let me try and remove the arduino-esp32 libs and see if it still fails ... here is the last compile which is a link step
```
xtensa-esp32-elf-gcc -Winline  -Og  -Wl,--gc-sections -L/home/kolban/esp32/Espruino/esp-idf/lib -L/home/kolban/esp32/Espruino/esp-idf/ld -L/home/kolban/esp32/Espruino/esp-idf/components/bt/lib -L/home/kolban/esp32/Espruino/esp-idf/components/esp32/lib -L/home/kolban/esp32/Espruino/template/build/bootloader -L/home/kolban/esp32/Espruino/template/build/bt -L/home/kolban/esp32/Espruino/template/build/driver -L/home/kolban/esp32/Espruino/template/build/esp32 -L/home/kolban/esp32/Espruino/template/build/esptool_py -L/home/kolban/esp32/Espruino/template/build/expat -L/home/kolban/esp32/Espruino/template/build/freertos -L/home/kolban/esp32/Espruino/template/build/json -L/home/kolban/esp32/Espruino/template/build/log -L/home/kolban/esp32/Espruino/template/build/lwip -L/home/kolban/esp32/Espruino/template/build/mbedtls -L/home/kolban/esp32/Espruino/template/build/newlib -L/home/kolban/esp32/Espruino/template/build/nghttp -L/home/kolban/esp32/Espruino/template/build/nvs_flash -L/home/kolban/esp32/Espruino/template/build/partition_table -L/home/kolban/esp32/Espruino/template/build/spi_flash -L/home/kolban/esp32/Espruino/template/build/tcpip_adapter -L/home/kolban/esp32/Espruino/template/build/arduino-esp32 -nostdlib -u call_user_start_cpu0 -Wl,--gc-sections -Wl,-static -Wl,-EL -o espruino_esp32.elf -Wl,--start-group  -lm -T esp32_out.ld -T/home/kolban/esp32/Espruino/esp-idf/components/esp32/ld/esp32.common.ld -T/home/kolban/esp32/Espruino/esp-idf/components/esp32/ld/esp32.rom.ld -T/home/kolban/esp32/Espruino/esp-idf/components/esp32/ld/esp32.peripherals.ld /home/kolban/esp32/Espruino/esp-idf/components/newlib/lib/libc.a /home/kolban/esp32/Espruino/esp-idf/components/newlib/lib/libm.a -lbt -lbtdm_app -ldriver -lesp32 /home/kolban/esp32/Espruino/esp-idf/components/esp32/libhal.a -lcrypto -lcore -lnet80211 -lphy -lrtc -lpp -lwpa -lexpat -lfreertos -ljson -llog -llwip -lmbedtls -lnghttp -lnvs_flash -lspi_flash -ltcpip_adapter -larduino-esp32 -lgcc src/jslex.o src/jsvar.o src/jsvariterator.o src/jsutils.o src/jsnative.o src/jsparse.o src/jspin.o src/jsinteractive.o src/jsdevices.o src/jstimer.o src/jsi2c.o src/jsspi.o src/jshardware_common.o /home/kolban/esp32/Espruino/Espruino/gen/jswrapper.o libs/compression/heatshrink/heatshrink_encoder.o libs/compression/heatshrink/heatshrink_decoder.o libs/compression/compress_heatshrink.o libs/network/network.o libs/network/socketserver.o libs/network/socketerrors.o libs/network/js/network_js.o libs/network/esp32/network_esp32.o targets/esp32/jshardware.o /home/kolban/esp32/Espruino/Espruino/gen/jspininfo.o src/jswrap_array.o src/jswrap_arraybuffer.o src/jswrap_date.o src/jswrap_error.o src/jswrap_espruino.o src/jswrap_flash.o src/jswrap_functions.o src/jswrap_interactive.o src/jswrap_io.o src/jswrap_json.o src/jswrap_modules.o src/jswrap_pin.o src/jswrap_number.o src/jswrap_object.o src/jswrap_onewire.o src/jswrap_pipe.o src/jswrap_process.o src/jswrap_promise.o src/jswrap_serial.o src/jswrap_spi_i2c.o src/jswrap_stream.o src/jswrap_string.o src/jswrap_waveform.o libs/math/jswrap_math.o libs/network/jswrap_net.o libs/network/http/jswrap_http.o libs/network/js/jswrap_jsnetwork.o libs/network/esp32/jswrap_esp32_network.o targets/esp32/jswrap_esp32.o targets/esp32/main.o   -Wl,--end-group

```

nkolban, 2016-10-27:
If I was to make a wild guess ... the issue is here:

https://github.com/espressif/esp-idf/blob/master/components/freertos/include/freertos/portable.h#L195

We recently changed this function to be inline ... but can we have inline function definitions in a header file and NOT get duplicate definitions?

Spritetm, 2016-10-27:
The function is in the header because we are basically using the inline function as a more pretty macro. The idea is that gcc inlines it every time it sees it. I think I made a mistake there trying to remember the inline rules... Can you try changing them into 'static inline' and rebuilding?

nkolban, 2016-10-27:
I changed the inlines to "static inlines" and all compiled cleanly and seems to run just fine.

Spritetm, 2016-10-27:
Okay, then I'll change that in master as well.

igrr, 2016-10-27:
Fixed in master

closed 2016-10-27 by igrr