espressif/esp-idf #63: Guru Meditation Error of type InstrFetchProhibited occurred on core

Xiben, 2016-10-27:
ESP32  keep rebooting

Output is like so: 

ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:920
load:0x40078000,len:2836
ho 0 tail 12 room 4
load:0x40098000,len:692
entry 0x4009813c
I (361) heap_alloc_caps: Initializing heap allocator:
I (362) heap_alloc_caps: Region 19: 3FFB4350 len 0002BCB0 tag 0
I (371) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (432) cpu_start: Pro cpu up.
I (466) cpu_start: Starting app cpu, entry point is 0x4008096c
I (0) cpu_start: App cpu up.
I (559) cpu_start: Pro cpu start user code
rtc v112 Sep 26 2016 22:32:10
XTAL 40M
I (739) cpu_start: Starting scheduler on PRO CPU.
I (214) cpu_start: Starting scheduler on APP CPU.
myaap ok!
Guru Meditation Error of type InstrFetchProhibited occurred on core   0. Except.
Register dump:
PC              :  00000000  PS        :  00060e30  A0           :  80059391  A1            :  3ffb3 
A2              :  3ffd0224  A3         :  3ffd01cc  A4             :  3ffd050d  A5             :  00000 
A6              :  3ffd0460  A7         :  00000001  A8           :  8000be88  A9            :  3ffb3 
A10            :  3ffd0224  A11       :  3ffd01cc  A12           :  3ffd04ec  A13            :  00000 
A14            :  00000001  A15      :  00000000  SAR        :  00000019  EXCCAUSE:  00000 
EXCVADDR:  00000000  LBEG    :  4000c2e0  LEND      :  4000c2f6  LCOUNT     :  00000 
Rebooting...

igrr, 2016-10-27:
Can you please upload application code which triggers this issue?

Xiben, 2016-10-28:
i feel because it's too many loops in a function.

Spritetm, 2016-10-28:
In that case, can you give us a bit of example code where too many loops in a function triggers this behaviour? Regardless of what it is, if we cannot replicate the issue, helping you is going to be pretty hard.

igrr, 2016-11-07:
Closing this because there doesn't seem to be any feedback. Please feel free to reopen if the issue still happens and you can provide the code to reproduce it.

closed 2016-11-07 by igrr