espressif/esp-idf #64: Execution locks up on 2016-10-28 builds of code ...

nkolban, 2016-10-28:
I downloaded the latest ESP-IDF and a virgin app template.  I added a couple of printlns and logs in my code.  All compiled cleanly and I flashed to my ESP32.  When I run the app, it locks up at boot.  Here is the output:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3ffc0000,len:0
load:0x3ffc0000,len:920
load:0x40078000,len:2836
ho 0 tail 12 room 4
load:0x40098000,len:692
entry 0x4009813c
I (80) heap_alloc_caps: Initializing heap allocator:
I (80) heap_alloc_caps: Region 19: 3FFC047C len 0001FB84 tag 0
I (81) heap_alloc_caps: Region 25: 3FFE8000 len 00018000 tag 1
I (91) cpu_start: Pro cpu up.
I (97) cpu_start: Single core mode
I (103) cpu_start: Pro cpu start user code
rtc v112 Sep 26 2016 22:32:10
XTAL 40M
I (140) cpu_start: Starting scheduler on PRO CP�
```

here is the app I am using:
```
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
static char tag[]= "tag";

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}

int app_main(void)
{
	ESP_LOGI(tag, "Starting");
    nvs_flash_init();
    system_init();
    tcpip_adapter_init();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    wifi_config_t sta_config = {
        .sta = {
            .ssid = "access_point_name",
            .password = "password",
            .bssid_set = false
        }
    };
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &sta_config) );
//    ESP_ERROR_CHECK( esp_wifi_start() );
//    ESP_ERROR_CHECK( esp_wifi_connect() );

    gpio_set_direction(GPIO_NUM_4, GPIO_MODE_OUTPUT);
    int level = 0;
    while (true) {
	ESP_LOGI(tag, "tick");
        gpio_set_level(GPIO_NUM_4, level);
        level = !level;
        vTaskDelay(300 / portTICK_PERIOD_MS);
printf("Tick\n");
    }

    return 0;
}
```

igrr, 2016-10-28:
Could you please mention two things: git commit of ESP-IDF used (lots of changes landed, not sure at which point you have got the code), and the contents of your sdkconfig file. Thanks.

nkolban, 2016-10-28:
The ESP-IDF commit is https://github.com/espressif/esp-idf/commit/19c4996bc87e460177a98643f02b40b587db040f

The esp-idf-template commit is https://github.com/espressif/esp-idf-template/commit/bbf4cd72594ab0ccc5c3bb7fed8b891196a576a1

My sdkconfig is attached at the end.  I don't have any JTAG technology nor know about it.  For the future, if we have a recipe for using JTAG that you can teach me, I'll write that up and teach others ... but right now I have nothing.   If ever needed, I can screen share live at any time if you need to run some live tests.  I can always be found on skype at neil.kolban

```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"

#
# Bootloader config
#
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=y
# CONFIG_LOG_BOOTLOADER_LEVEL_INFO is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=2

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
# CONFIG_ESPTOOLPY_COMPRESSED is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"

#
# Partition Table
#
CONFIG_PARTITION_TABLE_SINGLE_APP=y
# CONFIG_PARTITION_TABLE_TWO_OTA is not set
# CONFIG_PARTITION_TABLE_CUSTOM is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
CONFIG_PARTITION_TABLE_FILENAME="partitions_singleapp.csv"
CONFIG_APP_OFFSET=0x10000
CONFIG_OPTIMIZATION_LEVEL_DEBUG=y
# CONFIG_OPTIMIZATION_LEVEL_RELEASE is not set

#
# Component config
#
CONFIG_BT_RESERVE_DRAM=0

#
# ESP32-specific config
#
# CONFIG_ESP32_DEFAULT_CPU_FREQ_80 is not set
# CONFIG_ESP32_DEFAULT_CPU_FREQ_160 is not set
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
CONFIG_ESP32_ENABLE_STACK_WIFI=y
# CONFIG_ESP32_ENABLE_STACK_BT is not set
# CONFIG_ESP32_ENABLE_STACK_NONE is not set
CONFIG_MEMMAP_SMP=y
# CONFIG_MEMMAP_TRACEMEM is not set
CONFIG_TRACEMEM_RESERVE_DRAM=0x0
CONFIG_WIFI_ENABLED=y
CONFIG_SYSTEM_EVENT_QUEUE_SIZE=32
CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_MAIN_TASK_STACK_SIZE=4096
CONFIG_NEWLIB_STDOUT_ADDCR=y
# CONFIG_ULP_COPROC_ENABLED is not set
CONFIG_ULP_COPROC_RESERVE_MEM=0
CONFIG_ESP32_PANIC_PRINT_HALT=y
# CONFIG_ESP32_PANIC_PRINT_REBOOT is not set
# CONFIG_ESP32_PANIC_SILENT_REBOOT is not set
# CONFIG_ESP32_PANIC_GDBSTUB is not set
CONFIG_ESP32_DEBUG_OCDAWARE=y
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=10
CONFIG_TASK_WDT=y
# CONFIG_TASK_WDT_PANIC is not set
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK=y

#
# FreeRTOS
#
CONFIG_FREERTOS_UNICORE=y
CONFIG_FREERTOS_CORETIMER_0=y
# CONFIG_FREERTOS_CORETIMER_1 is not set
# CONFIG_FREERTOS_CORETIMER_2 is not set
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE is not set
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_PTRVAL=y
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY is not set
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=3
CONFIG_FREERTOS_ASSERT_FAIL_ABORT=y
# CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE is not set
# CONFIG_FREERTOS_ASSERT_DISABLE is not set
# CONFIG_ENABLE_MEMORY_DEBUG is not set
# CONFIG_FREERTOS_DEBUG_INTERNALS is not set

#
# Log output
#
# CONFIG_LOG_DEFAULT_LEVEL_NONE is not set
# CONFIG_LOG_DEFAULT_LEVEL_ERROR is not set
# CONFIG_LOG_DEFAULT_LEVEL_WARN is not set
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
# CONFIG_LOG_DEFAULT_LEVEL_DEBUG is not set
# CONFIG_LOG_DEFAULT_LEVEL_VERBOSE is not set
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_COLORS=y

#
# LWIP
#
CONFIG_LWIP_MAX_SOCKETS=4
CONFIG_LWIP_THREAD_LOCAL_STORAGE_INDEX=0
# CONFIG_LWIP_SO_REUSE is not set

#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=16384
# CONFIG_MBEDTLS_DEBUG is not set

#
# SPI Flash driver
#
# CONFIG_SPI_FLASH_ENABLE_COUNTERS is not set
```

nkolban, 2016-10-28:
Comments on the `#ESP32 IRC` chatgroup seem to echo that something is broken.

negativekelvin, 2016-10-28:
Did you reflash everything or just app-flash?

nkolban, 2016-10-28:
I re-flashed everything.  I also used the esptool erase_flash option to see if there was anything "lurking".  Unfortunately all tests result in the same story.  I am very open to the possibility that something I'm doing is wrong ... but as of now, I'm following the same recipes I had previously been following since I got my first ESP32 a few weeks back.  In summary ... am using Ubuntu on an x86-64 with the toolchain that I downloaded a few weeks back from Espressif.  I am using the latest (as of now/today) ESP-IDF and app-template.  I have no compilation errors, no flash issues ... just that when it runs ... it seems to stop before running my code.  And when I say "stop", I mean that output no longer appears on the serial output ... I have no idea what is going on internally at that point.

nkolban, 2016-10-28:
Issue resolved ... a colleague on `#ESP32 IRC` suggested that maybe I was using some form of mixed environment.  I thought to myself that couldn't be because I had removed all traces of the ESP-IDF and template app before downloading them again.  However, the failure was solidly on my part.  I had failed dismally to remember that my IDF_PATH was **not** pointing to the current/new download but was pointing to an older download.  As such, I believe I was mixing libraries and hence the mess.

I apologies to those who spent time on this issue when the issue was always at my end.  From this I learned to validate the value of IDF_PATH in addition to other procedures.

closed 2016-10-28 by nkolban