espressif/esp-idf #68: panic handler behaviour (Invoke GDBStub) link error

bvernoux, 2016-11-01:
I have updated esp-idf to latest commit (from today nov 1, 2016 but IIRC I had this problems 3 days ago too) and when I build esp-idf/examples/02_blink with make menuconfig => Component config => ESP32-specific config => panic handler behaviour (Invoke GDBStub)
I have a link error (panic.c => esp_gdbstub_panic_handler which is not referenced from )
```
make[1]: Leaving directory `/esp-idf/examples/02_blink/build/main'
LD blink.elf
/esp-idf/examples/02_blink/build/esp32/libesp32.a(panic.o):(.literal.commonErrorHandler+0x20): undefined reference to `esp_gdbstub_panic_handler'
/esp-idf/examples/02_blink/build/esp32/libesp32.a(panic.o): In function `commonErrorHandler':
/esp-idf/components/esp32/./panic.c:271: undefined reference to `esp_gdbstub_panic_handler'
collect2: error: ld returned 1 exit status
make: *** [/esp-idf/examples/02_blink/build/blink.elf] Error 1
```

projectgus, 2016-11-01:
Hi @bvernoux,

Thanks for reporting this. The fix you have made is the correct one. @Spritetm has a patch for this in our internal review queue, it should be merged very soon.

Angus

projectgus, 2016-11-02:
Should be fixed now, as of 8c1d1e1

Thanks,

Angus

closed 2016-11-02 by projectgus