espressif/esp-idf #69: EAP support?

ivanO86, 2016-11-02:
Hi all,

I have just started a new project with an ESP32 board and I need WPA-enterprise support for some applications. It is not a mandatory feature at this moment, but I will need it, and I can not find a solution in your esp_wifi library. Are you thinking about add this encryption support?

Thank you. Best regards

negativekelvin, 2016-11-02:
> WiFi WPA2-Enterprise and WPS are not supported **yet**

https://github.com/espressif/esp-idf/releases

igrr, 2016-11-07:
WPS support is going to be added soon (~this week). WPA-enterprise is going to be added somewhere before 1.0 release at the end of this month.

ivanO86, 2016-11-07:
Thanks for your update.

projectgus, 2016-12-22:
WPA Enterprise now supports EAP-TLS, PEAP-MSCHAPv2 and TTLS-MSCHAPv2.

closed 2016-12-22 by projectgus