espressif/esp-idf #73: Link the libsmartconfig.a failed

icamgo, 2016-11-03:
............
............
make[1]: Leaving directory '/work/xwifi/esp-idf/examples/07_smartconfig/build/main'
LD smartconfig-demo.elf
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o):(.literal.smartconfig_done+0x28): undefined reference to `esp_smartconfig_stop'
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o):(.literal.smartconfig_task+0x8): undefined reference to `esp_smartconfig_start'
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o):(.literal.smartconfig_init+0x8): undefined reference to `esp_smartconfig_set_type'
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o): In function `smartconfig_done':
/work/xwifi/esp-idf/examples/07_smartconfig/main/./app_main.c:65: undefined reference to `esp_smartconfig_stop'
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o): In function `smartconfig_task':
/work/xwifi/esp-idf/examples/07_smartconfig/main/./app_main.c:65: undefined reference to `esp_smartconfig_start'
/work/xwifi/esp-idf/examples/07_smartconfig/build/main/libmain.a(app_main.o): In function `smartconfig_init':
/work/xwifi/esp-idf/examples/07_smartconfig/main/./app_main.c:93: undefined reference to `esp_smartconfig_set_type'
collect2: error: ld returned 1 exit status
/work/xwifi/espidf/make/project.mk:214: recipe for target '/work/xwifi/esp-idf/examples/07_smartconfig/build/smartconfig-demo.elf' failed
make: *** [/work/xwifi/esp-idf/examples/07_smartconfig/build/smartconfig-demo.elf] Error 1
.............

igrr, 2016-11-03:
I'm not able to reproduce the issue here. Can you please upload output of the following commands to a gist and post link here?
```
git status
git submodule foreach 'git status'
make clean
make V=1
```
Thanks.

icamgo, 2016-11-03:
I have pushed the example to : https://github.com/icamgo/esp-idf/tree/noduino/examples/07_smartconfig/
Using the latest esp-idf, esp32/lib is (should be the latest one):

comcat@jackslab:/work/xwifi/esp-idf/components/esp32/lib$ git status
HEAD detached at b3090d8
nothing to commit, working directory clean

igrr, 2016-11-03:
I am able to build smartconfig branch with the latest master without errors.
Please post the output of the commands mentioned above in a gist :)

icamgo, 2016-11-03:
https://gist.github.com/icamgo/9e297e4b1aad0d838b619e77e75a86c0

igrr, 2016-11-03:
Your `IDF_PATH` points to `/work/xwifi/espidf`, while you are running everything in `/work/xwifi/esp-idf`. You may need to `export IDF_PATH=/work/xwifi/esp-idf`.

icamgo, 2016-11-03:
This is .... Build system can compute the IDF_PATH automaticlly. Something like the implemetation of the build system in our esp8266/esp8285 sdk enviroment: http://github.com/icamgo/noduino-sdk

igrr, 2016-11-03:
This is indeed possible for examples which are in-tree, using relative paths (like you do [here](https://github.com/icamgo/noduino-sdk/blob/master/examples/rawsdk/adc/Makefile#L16)). However if the example is copied to a random directory outside of ESP-IDF tree, this relative path will no longer work. The primary use case for ESP-IDF is to be used with applications developed outside of ESP-IDF tree, so for this reason we suggest that developers set `IDF_PATH` either in the environment, or in application makefile.

closed 2016-11-03 by igrr