espressif/esp-idf #74: esptool.py compatibility with python>=3

endeebee, 2016-11-03:
Upon compiling the hello world example successfully, esptool.py has trouble running because my Python version is 3.5.
Upon checking the source, I saw that some print statements use parentheses and some others do not. I have changed all the print statements to be compatible with Python 3, however I am still getting an error:

>A fatal error occurred: /.../bootloader.elf  has invalid ELF magic header

May this also be a problem with the version of Python used?

projectgus, 2016-11-03:
Hi @endeebee,

At the moment esptool.py only supports Python 2. There is an effort underway to support both Python 2 & 3, but we aren't there yet. Sorry!

You can use the menuconfig option "SDK Tool Configuration" -> "Python 2 interpreter" to specify the name of the Python 2 interpreter on your system.

Angus

closed 2016-11-03 by projectgus

krzychb, 2016-11-04:
Hi @projectgus,

Information aboout python version was obvous to me basing on previus experience with ESP8266. In documentation for ESP32 python is bearly mentioned and I do not see it at all in windows-setup.rst. 

To  make it easier for newcomers, I propose to update deocumentation. Aa s minimum I would include information what Python version is supported in:

- windows-setup.rst
- macos-setup.rst
- linux-setup.rst

Krzysztof

endeebee, 2016-11-04:
@projectgus   Thanks, that did the trick!
@krzychb I initially tried changing the shebang in esptool.py, which didn't work, which lead me to convert the whole script to python3, which also didn't work. I thus agree that it might be beneficial to mention in the documentation which settings should definitely be made with make menuconfig and which are only optional for a working compile environment.

Nik