espressif/esp-idf #76: switching vprintf to uart1

lollizze, 2016-11-04:
Hi!
I recently updated my esp-idf to the latest and then i ran into some issues as expected because i was way behind, but there is one thing that i can't find a solution on. Before it was possible to use `uart_tx_switch(1) `to change `vprintf `to uart1, now the only print function that prints to uart1 is `ets_printf`. 

Is there a way to change printf to uart1, or is it a bug that uart_tx_switch does not do it?

nkolban, 2016-11-04:
If we look at the excellent code in the Arduino HAL functions ... specifically here ...

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L329

We see a way that the Arduino HAL has achieved the technique.  Remember that you can use the Arduino libraries in your ESP-IDF solutions.

There is a great forum for ESP32 found here ... http://esp32.com/index.php  ... you may find you get answers quicker there.

igrr, 2016-11-04:
@lollizze Does `freopen("/dev/uart/1", "w", stdout);` do the trick? (UART1 needs to be initialized).

lollizze, 2016-11-04:
@nkolban not really the same, i would like to use vprintf not implement my own

@igrr I will try your suggestion but shouldnt uart_tx_switch do that?

negativekelvin, 2016-11-04:
@lollizze uart_tx_switch and ets_printf are ROM functions not thread safe for calling from rtos environment.  This is where the UART is initialized in rtos context:  https://github.com/espressif/esp-idf/blob/abecab7525e7edb1fde16ab5d8cf7b368b1d332c/components/esp32/cpu_start.c#L176

projectgus, 2016-12-22:
I'm going to close this, because between uart_tx_switch (for ROM functions) and freopen (for newlib libc stdout), it should be possible to remap stdout.

There is also now a config option under Component Config -> ESP32-specific Config for "UART for console output". This allows you to change the default UART used for console after reset.

Please reopen if any of these features do not work as expected.

closed 2016-12-22 by projectgus