espressif/esp-idf #78: VFS: Return of -1 from VFS open call results in a file descriptor of 4095

nkolban, 2016-11-06:
I am practicing working with VFS.  I returned a constant of "-1" from my open() implementation as a test and find that the resulting file descriptor returned is 4095.  I believe that the indication of an error (-1) is being ignored.

I created a VFS definition that looks like:

```
void registerTestVFS(char *mountPoint) {
	esp_vfs_t vfs;
	esp_err_t err;

	vfs.fd_offset = 0;
	vfs.flags = ESP_VFS_FLAG_DEFAULT;
	vfs.write  = vfs_write;
	vfs.lseek  = vfs_lseek;
	vfs.read   = vfs_read;
	vfs.open   = vfs_open;
	vfs.close  = vfs_close;
	vfs.fstat  = vfs_fstat;
	vfs.stat   = vfs_stat;
	vfs.link   = vfs_link;
	vfs.rename = vfs_rename;

	err = esp_vfs_register(mountPoint, &vfs, NULL);
	if (err != ESP_OK) {
		ESP_LOGE(tag, "esp_vfs_register: err=%d", err);
	}
} // End of registerTestVFS
```

For my vfs_open function, I coded the following:

```
static int vfs_open(const char *path, int flags, int accessMode) {
	ESP_LOGI(tag, ">> open path=%s, flags=0x%x, accessMode=0x%x", path, flags, accessMode);
	logFlags(flags);
	return -1;
}
```

I invoked it all with:
```
    registerTestVFS("/data");
    ESP_LOGI(tag, "vfs registered");

    int fd = open("/data/x", O_CREAT, 0x777);
    ESP_LOGD(tag, "fd from open = %d", fd);
```

The result was:
```
I (564) vfs-skeleton: vfs registered
I (567) vfsTest: >> open path=/x, flags=0x200, accessMode=0x777
D (572) vfsTest: flags:
D (574) vfsTest: - O_CREAT
D (577) vfs-skeleton: fd from open = 4095
```

igrr, 2016-11-06:
Adding the following check in `esp_vfs_open`, after `CHECK_AND_CALL` line should fix it:
```c++
    if (ret == -1) {
        return ret;
    }
```

nkolban, 2016-11-06:
Thank you sir.  I patched that in for a test and it seems to work great.

commit 80abb531 (2019-06-30): Merge branch 'feature/remove_mbedtls_cmake_workarounds' into 'master'

mbedtls: remove component CMake file workarounds

Closes #78

See merge request idf/esp-idf!5341
closed 2016-11-09 by igrr