espressif/esp-idf #81: Confusing about _times_r implementation and newlibc clock

jolivepetrus, 2016-11-07:
I'm a bit confusing about _times_r implementation.

Current implementation:

clock_t IRAM_ATTR _times_r(struct _reent *r, struct tms *ptms)
{
    clock_t t = xTaskGetTickCount() * (portTICK_PERIOD_MS * CLK_TCK / 1000);
    ptms->tms_cstime = t;
    ptms->tms_cutime = 0;
    ptms->tms_stime = t;
    ptms->tms_utime = 0;
    struct timeval tv = {0, 0};
    _gettimeofday_r(r, &tv, NULL);
    return (clock_t) tv.tv_sec;
}

In newlibc clock implementation is:

clock_t 
clock ()
{
  struct tms tim_s;
  clock_t res;

  if ((res = (clock_t) _times_r (_REENT, &tim_s)) != (clock_t) -1)
    res = (clock_t) (tim_s.tms_utime + tim_s.tms_stime +
		     tim_s.tms_cutime + tim_s.tms_cstime);

  return res;
}

It seems that ticks returned by clock are multiplied by 2, due to in _times_r tms_cstime and tms_stime are set with the current processor ticks. tms_cstime or tms_stime must be set to 0.

igrr, 2016-11-07:
Thanks, I haven't checked `clock`, will fix and add a test for it.

igrr, 2016-11-08:
Okay, i have found why i implemented `_times_r` incorrectly in the first place. I missed one 's' in this line of manpage related to `times` function:

```
tms_cstime  The sum of the tms_stimes and tms_cstimes of the child processes.
                                    ^
```

igrr, 2016-11-09:
Fixed in master

closed 2016-11-09 by igrr