espressif/esp-idf #82: spi_flash working?

jolivepetrus, 2016-11-07:
We are working in SPIFFS. Things go well, using SPIFFS in our Lua RTOS, with file system access from Lua. But some times we received exceptions, and I think that some dead locks running on multicore. It seems that not happens using uni core.

I try to reproduce this in the attached code (tested, I think that flash address are on my flash bounds). In our build we receive this exception:

Guru Meditation Error: Core   0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      :  40007c78  PS      :  00060e34  A0      :  80007d16  A1      :  3ffb3770  
A2      :  00800000  A3      :  3ffbaa20  A4      :  8008175c  A5      :  3ffbaa20  
A6      :  0000018d  A7      :  0000000d  A8      :  3ff40000  A9      :  000000d0  
A10     :  00800000  A11     :  3ff4001c  A12     :  8008170c  A13     :  3ffbaa00  
A14     :  00000000  A15     :  3ffae270  SAR     :  00000017  EXCCAUSE:  00000005  
EXCVADDR:  00000000  LBEG    :  00000000  LEND    :  00000000  LCOUNT  :  00000000  
Rebooting...

Sorry, I don't have more information about this, I haven't a JTAG for know more about this ... but something happens with spi_flash function and multicore ....

Sample code:

#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "esp_spi_flash.h"

void spi_flash_task(void *pvParameter)
{
	char buff[4096];
	int ret;

	uint32_t base = 0x00180000;
	uint32_t page = 4096;
	uint32_t size = 0x80000 - page;
	uint32_t addr = base;
	
	for(;;) {
		addr = base;
		
		while(addr < base + size - page) {		
			ret = spi_flash_read(addr, buff, page);
			if (ret != 0)
				printf("oops 1\r\n");

			ret = spi_flash_erase_sector(addr >> 12);
			if (ret != 0)
				printf("oops 2\r\n");
			
			addr += page;		
		}
	}
	
	vTaskDelete(NULL);
}

void app_main()
{
    nvs_flash_init();
    system_init();
    xTaskCreate(&spi_flash_task, "spiftask", 1024*10, NULL, 5, NULL);
}

igrr, 2016-11-07:
Thanks, I'll try to reproduce this tomorrow. Could you please attach sdkconfig file you have used for this test?

jolivepetrus, 2016-11-07:
Of course:

#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python"

#
# Bootloader config
#
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=y
# CONFIG_LOG_BOOTLOADER_LEVEL_INFO is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE is not set
CONFIG_LOG_BOOTLOADER_LEVEL=2

#
# Serial flasher config
#
CONFIG_ESPTOOLPY_PORT="/dev/tty.SLAB_USBtoUART"
# CONFIG_ESPTOOLPY_BAUD_115200B is not set
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
CONFIG_ESPTOOLPY_BAUD_921600B=y
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=921600
CONFIG_ESPTOOLPY_COMPRESSED=y
# CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_8MB is not set
# CONFIG_ESPTOOLPY_FLASHSIZE_16MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE="2MB"

#
# Partition Table
#
CONFIG_PARTITION_TABLE_SINGLE_APP=y
# CONFIG_PARTITION_TABLE_TWO_OTA is not set
# CONFIG_PARTITION_TABLE_CUSTOM is not set
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
CONFIG_PARTITION_TABLE_FILENAME="partitions_singleapp.csv"
CONFIG_APP_OFFSET=0x10000
CONFIG_OPTIMIZATION_LEVEL_DEBUG=y
# CONFIG_OPTIMIZATION_LEVEL_RELEASE is not set

#
# Component config
#
CONFIG_BT_RESERVE_DRAM=0

#
# ESP32-specific config
#
# CONFIG_ESP32_DEFAULT_CPU_FREQ_80 is not set
# CONFIG_ESP32_DEFAULT_CPU_FREQ_160 is not set
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ=240
CONFIG_ESP32_ENABLE_STACK_WIFI=y
# CONFIG_ESP32_ENABLE_STACK_BT is not set
# CONFIG_ESP32_ENABLE_STACK_NONE is not set
CONFIG_MEMMAP_SMP=y
# CONFIG_MEMMAP_TRACEMEM is not set
CONFIG_TRACEMEM_RESERVE_DRAM=0x0
CONFIG_WIFI_ENABLED=y
CONFIG_SYSTEM_EVENT_QUEUE_SIZE=32
CONFIG_SYSTEM_EVENT_TASK_STACK_SIZE=2048
CONFIG_MAIN_TASK_STACK_SIZE=4096
CONFIG_NEWLIB_STDOUT_ADDCR=y
# CONFIG_ULP_COPROC_ENABLED is not set
CONFIG_ULP_COPROC_RESERVE_MEM=0
# CONFIG_ESP32_PANIC_PRINT_HALT is not set
CONFIG_ESP32_PANIC_PRINT_REBOOT=y
# CONFIG_ESP32_PANIC_SILENT_REBOOT is not set
# CONFIG_ESP32_PANIC_GDBSTUB is not set
CONFIG_ESP32_DEBUG_OCDAWARE=y
CONFIG_INT_WDT=y
CONFIG_INT_WDT_TIMEOUT_MS=300
CONFIG_INT_WDT_CHECK_CPU1=y
CONFIG_TASK_WDT=y
# CONFIG_TASK_WDT_PANIC is not set
CONFIG_TASK_WDT_TIMEOUT_S=5
CONFIG_TASK_WDT_CHECK_IDLE_TASK=y
CONFIG_TASK_WDT_CHECK_IDLE_TASK_CPU1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_RTC_CLOCK_SOURCE_INTERNAL_RC=y

#
# FreeRTOS
#
# CONFIG_FREERTOS_UNICORE is not set
CONFIG_FREERTOS_CORETIMER_0=y
# CONFIG_FREERTOS_CORETIMER_1 is not set
# CONFIG_FREERTOS_CORETIMER_2 is not set
CONFIG_FREERTOS_HZ=100
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=y
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_NONE=y
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_PTRVAL is not set
# CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY is not set
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=1
CONFIG_FREERTOS_ASSERT_FAIL_ABORT=y
# CONFIG_FREERTOS_ASSERT_FAIL_PRINT_CONTINUE is not set
# CONFIG_FREERTOS_ASSERT_DISABLE is not set
CONFIG_FREERTOS_BREAK_ON_SCHEDULER_START_JTAG=y
# CONFIG_ENABLE_MEMORY_DEBUG is not set
CONFIG_FREERTOS_ISR_STACKSIZE=1536
# CONFIG_FREERTOS_DEBUG_INTERNALS is not set

#
# Log output
#
# CONFIG_LOG_DEFAULT_LEVEL_NONE is not set
# CONFIG_LOG_DEFAULT_LEVEL_ERROR is not set
# CONFIG_LOG_DEFAULT_LEVEL_WARN is not set
# CONFIG_LOG_DEFAULT_LEVEL_INFO is not set
# CONFIG_LOG_DEFAULT_LEVEL_DEBUG is not set
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=y
CONFIG_LOG_DEFAULT_LEVEL=5
CONFIG_LOG_COLORS=y

#
# LWIP
#
CONFIG_LWIP_MAX_SOCKETS=4
CONFIG_LWIP_THREAD_LOCAL_STORAGE_INDEX=0
# CONFIG_LWIP_SO_REUSE is not set
CONFIG_LWIP_DHCP_MAX_NTP_SERVERS=1

#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=16384
# CONFIG_MBEDTLS_DEBUG is not set

#
# SPI Flash driver
#
# CONFIG_SPI_FLASH_ENABLE_COUNTERS is not set

jolivepetrus, 2016-11-07:
Of course:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/575787/sdkconfig.txt)

igrr, 2016-11-08:
I think my flash chip now has a nice hole in this region, as the test is running for about an hour already. Unfortunately I haven't been able reproduce the issue so far.

jolivepetrus, 2016-11-08:
Thanks for your test. I will try to do more tests, and start the buy process for a new flash ship :) ... In my tests, exeption is raised after a few seconds to start it.

jolivepetrus, 2016-11-08:
Have you tested with the attached sdkconfig in the previous comments?
I have tested again and fails.
If I test with sdkconfig attached to this comment don't fail. This sdkconfig has the WDT disabled.

# CONFIG_INT_WDT is not set
# CONFIG_TASK_WDT is not set

jolivepetrus, 2016-11-08:
Have you tested with the attached sdkconfig in the previous comments?
I have tested again and fails.
If I test with sdkconfig attached to this comment don't fail. This sdkconfig has the WDT disabled.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/577883/sdkconfig.txt)

igrr, 2016-11-08:
Yes, i have tested with sdkconfig you have provided above.
Which commit of ESP-IDF have you been testing this with?

jolivepetrus, 2016-11-08:
This is the comit (git rev-parse HEAD):

c8a43508e5a08737b5e4eb0c1e43a374cb84ddb7

jolivepetrus, 2016-11-08:
Sorry, I know that this is not suitable for this, but I need email to some one of the esp-idf developers. I think that we are working in an interesting thing and maybe part of our work will be useful for you. Our work have key parts that need FreeRTOS changes ...

Here is my email: jolive@iberoxarxa.com

jolivepetrus, 2016-11-16:
I think that this issue is related with something wrong with one of our UART interrupt handler, where the following code is running, and intensive spi_flash access (read) is done from another task. When interrupt is processed, then we give panic.

....
....
compute byte calling functions on IRAM
....
....
 xQueueSendFromISR(q, &byte, &xHigherPriorityTaskWoken);
.....
.....
if (xHigherPriorityTaskWoken) {
    portYIELD_FROM_ISR();
}

I'm trying to set up JTAG. I hope to provide more information about this in the following days.

jolivepetrus, 2016-11-17:
This issue was caused because we don't crear the interrupt flag on the right place in our UART interrupt handler.

Sorry for the inconvenience.

closed 2016-11-17 by jolivepetrus