espressif/esp-idf #83: Undefined reference to __ieee754_remainder

nkolban, 2016-11-07:
I am compiling some large chunks of foreign code and my linkage fails with the following error:
```
/home/kolban/esp32/esptest/esp-idf/components/newlib/lib/libm.a(lib_a-e_fmod.o):(.literal+0x0): undefined reference to `__ieee754_remainder'
/home/kolban/esp32/esptest/esp-idf/components/newlib/lib/libm.a(lib_a-e_fmod.o): In function `__ieee754_fmod':
/home/ivan/freertos/smpemu/newlib/newlib_xtensa-2.2.0/xtensa-esp108-elf/newlib/libm/math/../../../.././newlib/libm/math/e_fmod.c:35: undefined reference to `__ieee754_remainder'
```

Digging down, it seems that this function is used in newlib/libm (see: https://github.com/espressif/newlib-esp32/blob/4ddc99d47677b94c115cafcaa4a6939042ea8c4f/newlib/libm/math/e_fmod.c#L38)

while the implementation of it provided by newlib is commented out

https://github.com/espressif/newlib-esp32/blob/4ddc99d47677b94c115cafcaa4a6939042ea8c4f/newlib/libm/math/e_remainder.c#L29

There is a comment that says "This is provided by libgcc" but I am using ESP-IDF and haven't added or removed any libraries from the build.  So out of the box, something would appear to need to be changed.

svaarala, 2016-11-26:
Possibly related: https://github.com/espressif/esp-idf/issues/83.

nkolban, 2016-11-26:
I have found my first impact on this issue.  see... https://github.com/svaarala/duktape/issues/1107

closed 2016-12-13 by igrr

DavidRimon, 2017-05-28:
if the  solution above doesn't work, this worked for me while trying to compile GPS_HardwareSerial_parsion.ino from Adafruit GPS (which I was using for NodeMCU 0.9)

double fmod(double x, double y) {
	return x - y * floor(x / y);
}

wongjiajie, 2018-01-14:
Hey DavideRimon can you share the code with me for the nodemcu and adafruit gps? Thanks alot