espressif/esp-idf #84: Eclipse

forthlightning, 2016-11-08:
I have followed the macos setup guide and flashed code to the board successfully using the command line tool.

Subsequently I have tried to get eclipse set up, changing defaults as described in the eclipse setup guide. I am getting the following output when I try to build.

`make all 
including /Users/maxfieldmorrison/esp/esp-idf/components/bootloader/Makefile.projbuild...
including /Users/maxfieldmorrison/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /Users/maxfieldmorrison/esp/esp-idf/components/mbedtls/Makefile.projbuild...
including /Users/maxfieldmorrison/esp/esp-idf/components/nghttp/Makefile.projbuild...
including /Users/maxfieldmorrison/esp/esp-idf/components/partition_table/Makefile.projbuild...
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/maxfieldmorrison/esp/esp-idf/components/bootloader/src V=1 SDKCONFIG=/Users/maxfieldmorrison/esp/myapp/build/bootloader/sdkconfig BUILD_DIR_BASE=/Users/maxfieldmorrison/esp/myapp/build/bootloader  /Users/maxfieldmorrison/esp/myapp/build/bootloader/bootloader.bin
including /Users/maxfieldmorrison/esp/esp-idf/components/esptool_py/Makefile.projbuild...
including /Users/maxfieldmorrison/esp/esp-idf/components/bootloader/Makefile.projbuild...
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/maxfieldmorrison/esp/myapp/build/bootloader/log -f /Users/maxfieldmorrison/esp/esp-idf/components/log/component.mk COMPONENT_BUILD_DIR=/Users/maxfieldmorrison/esp/myapp/build/bootloader/log build
Target 'log-build' responsible for '/Users/maxfieldmorrison/esp/myapp/build/bootloader/log/liblog.a'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/maxfieldmorrison/esp/myapp/build/bootloader/spi_flash -f /Users/maxfieldmorrison/esp/esp-idf/components/spi_flash/component.mk COMPONENT_BUILD_DIR=/Users/maxfieldmorrison/esp/myapp/build/bootloader/spi_flash build
Target 'spi_flash-build' responsible for '/Users/maxfieldmorrison/esp/myapp/build/bootloader/spi_flash/libspi_flash.a'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/maxfieldmorrison/esp/myapp/build/bootloader/main -f /Users/maxfieldmorrison/esp/esp-idf/components/bootloader/src/main/component.mk COMPONENT_BUILD_DIR=/Users/maxfieldmorrison/esp/myapp/build/bootloader/main build
Target 'main-build' responsible for '/Users/maxfieldmorrison/esp/myapp/build/bootloader/main/libmain.a'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C /Users/maxfieldmorrison/esp/myapp/build/bt -f /Users/maxfieldmorrison/esp/esp-idf/components/bt/component.mk COMPONENT_BUILD_DIR=/Users/maxfieldmorrison/esp/myapp/build/bt build
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -MMD -MP -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -Wno-error=unused-label -Wno-error=return-type -Wno-error=missing-braces -Wno-error=pointer-sign -Wno-error=parentheses -DESP_PLATFORM  -I /Users/maxfieldmorrison/esp/esp-idf/components/bt/include -I /Users/maxfieldmorrison/esp/esp-idf/components/driver/include -I /Users/maxfieldmorrison/esp/esp-idf/components/esp32/include -I /Users/maxfieldmorrison/esp/esp-idf/components/expat/port/include -I /Users/maxfieldmorrison/esp/esp-idf/components/expat/include/expat -I /Users/maxfieldmorrison/esp/esp-idf/components/freertos/include -I /Users/maxfieldmorrison/esp/esp-idf/components/json/include -I /Users/maxfieldmorrison/esp/esp-idf/components/json/port/include -I /Users/maxfieldmorrison/esp/esp-idf/components/log/include -I /Users/maxfieldmorrison/esp/esp-idf/components/lwip/include/lwip -I /Users/maxfieldmorrison/esp/esp-idf/components/lwip/include/lwip/port -I /Users/maxfieldmorrison/esp/esp-idf/components/lwip/include/lwip/posix -I /Users/maxfieldmorrison/esp/esp-idf/components/mbedtls/port/include -I /Users/maxfieldmorrison/esp/esp-idf/components/mbedtls/include -I /Users/maxfieldmorrison/esp/esp-idf/components/newlib/include -I /Users/maxfieldmorrison/esp/esp-idf/components/newlib/platform_include -I /Users/maxfieldmorrison/esp/esp-idf/components/nghttp/port/include -I /Users/maxfieldmorrison/esp/esp-idf/components/nghttp/include -I /Users/maxfieldmorrison/esp/esp-idf/components/nvs_flash/include -I /Users/maxfieldmorrison/esp/esp-idf/components/openssl/include -I /Users/maxfieldmorrison/esp/esp-idf/components/spi_flash/include -I /Users/maxfieldmorrison/esp/esp-idf/components/tcpip_adapter/include -I /Users/maxfieldmorrison/esp/esp-idf/components/vfs/include -I /Users/maxfieldmorrison/esp/esp-idf/components/wpa_supplicant/include -I /Users/maxfieldmorrison/esp/esp-idf/components/wpa_supplicant/port/include -I /Users/maxfieldmorrison/esp/esp-idf/components/xtensa-debug-module/include -I /Users/maxfieldmorrison/esp/myapp/main/include -I /Users/maxfieldmorrison/esp/myapp/build/include  -I. -c /Users/maxfieldmorrison/esp/esp-idf/components/bt/./bt.c -o bt.o
make[1]: xtensa-esp32-elf-gcc: No such file or directory
make[1]: *** [bt.o] Error 1
make: *** [bt-build] Error 2`

The problems tab has an error with the following text:

`Description	Resource	Path	Location	Type
Program "xtensa-esp32-elf-gcc" not found in PATH	myapp		Project Properties, C++ Preprocessor Include.../Providers, [CDT Cross GCC Built-in Compiler Settings] options	C/C++ Scanner Discovery Problem
`

Any help would be much appreciated!

nkolban, 2016-11-08:
In order to compile C source code for execution on an ESP32 you need a "cross compiler".  This is a tool that runs on your compilation platform (either Windows or Linux) and generates executable code that will run on the ESP32.  The cross compiler executable is called "xtensa-esp32-elf-gcc".

If you are building on Linux, read and follow these instructions:

http://esp-idf.readthedocs.io/en/latest/linux-setup.html

If you are building on Windows, read and follow these instructions:

http://esp-idf.readthedocs.io/en/latest/windows-setup.html

forthlightning, 2016-11-08:
Yes I have made the toolchain and can compile using the command line "make flash."

It appears that Eclipse cannot find this cross compiler executable. 

I have confirmed that eclipse can see my PATH variable, and that the PATH includes the cross compiler location.

This is the output from echo $PATH

`/Library/Frameworks/Python.framework/Versions/2.7/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:/Users/maxfieldmorrison/esp/xtensa-esp32-elf/bin:/Users/maxfieldmorrison/esp/xtensa-esp32-elf/bin`

nkolban, 2016-11-08:
I run Eclipse in both Windows and Linux and my experience has been that Eclipse uses the PATH environment in which it runs to find commands/executables.  I am NOT a Mac user I'm afraid so I can't speak to global environments on that platform.  On Windows, one would edit the master environment variables to add the xtensa-esp32-elf/bin onto PATH.  In Linux, I add that in my .profile shell script to add to PATH.

At a gut guess, I would say that when you launch Eclipse from your desktop, it is not picking up the content of your user's ".profile" file.  Can you try and launch Eclipse by hand from a terminal window in which you absolutely know that the environment variable is set properly?

I did a google search and found this ...

http://stackoverflow.com/questions/7501678/set-environment-variables-on-mac-os-x-lion

Read the section on "environment.plist".  That looks like a direct match.

igrr, 2016-11-09:
I believe the recommended approach is to set PATH variable in your project/workspace preferences.

forthlightning, 2016-11-11:
Ok thanks so much! 

So I updated the path in the IDE project settings. Eclipse can now find the esp-idf toolchain. 

But now I'm getting the following error:

`File "/Users/maxfieldmorrison/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 25, in <module>
    import serial
ImportError: No module named serial
make: *** [flash] Error 1`

This is my current PATH in the IDE:

`/bin:/usr/bin:/usr/sbin:/sbin:/Users/myname/esp/esp-idf:/usr/bin/python`

I definitely have serial module in python. I have navigated to the esptool folder and opened a python terminal and tried to `import serial` with success.

projectgus, 2016-12-22:
Sorry for the delayed response. Are you still seeing this issue?

It seems like in some circumstances Eclipse injects its own Python process onto the PATH, ahead of system Python. If you run `make menuconfig` in a terminal and set the Python path to /usr/bin/python then this will probably work around the issue. Although that's not a good long-term fix, please let us know if it solves the problem in this case.

projectgus, 2017-02-16:
@forthlightning did you make any more progress with this issue?

projectgus, 2017-03-22:
The key appears to be overriding PYTHONPATH. We've added a note to the docs on how to set this:
http://esp-idf.readthedocs.io/en/latest/eclipse-setup.html#project-properties

There may be a better solution in the future, but for now I'm going to close this issue.

closed 2017-03-22 by projectgus

SpiRaiL, 2020-11-17:
I solved it by simply installing python from the python website. Sounds dumb.. but it was a "Microsoft store" installation that eclipse was not able to find. Of course MS decided it had a better place to put the executable than the recommended install location of python.