espressif/esp-idf #86: Panic when "01_hello_world" example restarts

krzychb, 2016-11-09:
Hi,

When the standard [01_hello_world](https://github.com/espressif/esp-idf/blob/master/examples/01_hello_world/main/hello_world_main.c) example soft restarts, the following is shown on serial monitor:

```
...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
Restarting now.
C:/msys32/esp-idf/components/freertos/./queue.c:716 (xQueueGenericSend)- assert failed!
Guru Meditation Error: Core   0 panic'ed (Unhandled debug exception)
Register dump:
PC      :  40009203  PS      :  00060836  A0      :  800d06aa  A1      :  3ffb7ca0  
...
```

Is it expected behaviour?

If so, is there a reasonable way to soft restart ESP32 gracefully, without any panicking?

projectgus, 2016-11-15:
Hi @krzychb ,

Thanks for reporting this. It isn't expected behaviour, it's a bug in the system_restart() function. We're working on a fix internally, I'll update when it's available.

Angus

igrr, 2016-11-24:
@krzychb @jmattson this should be fixed in master now. If you are still seeing the problem, feel free to reopen.

closed 2016-11-24 by igrr

krzychb, 2016-11-25:
I see the problem fixed / on panic anymore.
Thanks!