espressif/esp-idf #93: Non determinitic panic (Interrupt wdt timeout on CPU0)

nkolban, 2016-11-12:
I have a relatively small application that doesn't use any WiFi ... it is only using the new UART library.  What I am finding is that when I boot my app, 1 time in 5 it crashes.  Here are the details:

```
I (130) cpu_start: Starting scheduler on PRO CPU.
D (131) gps: >> doGPS
Guru Meditation Error: Core   0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      :  400d09b6  PS      :  00060034  A0      :  40080ed9  A1      :  3ffb1560  
A2      :  3ffb4ad8  A3      :  00000004  A4      :  00000001  A5      :  00000004  
A6      :  00000000  A7      :  3ff50000  A8      :  00000007  A9      :  3ffb1540  
A10     :  00000000  A11     :  00000001  A12     :  00000078  A13     :  3ffb1568  
A14     :  3ffb4d98  A15     :  40084374  SAR     :  0000001f  EXCCAUSE:  00000005  
EXCVADDR:  00000000  LBEG    :  4000c2e0  LEND    :  4000c2f6  LCOUNT  :  00000000  
```

And using gdb we find
```
(gdb) list *0x400d09b6
0x400d09b6 is in esp_task_wdt_feed (/home/kolban/esp32/esptest/esp-idf/components/esp32/./task_wdt.c:114).
109	        //All tasks have checked in; time to feed the hw watchdog.
110	        TIMERG0.wdt_wprotect=TIMG_WDT_WKEY_VALUE;
111	        TIMERG0.wdt_feed=1;
112	        TIMERG0.wdt_wprotect=0;
113	        //Reset fed_watchdog status
114	        for (wdttask=wdt_task_list; wdttask->next!=NULL; wdttask=wdttask->next) wdttask->fed_watchdog=false;
115	    }
116	}
117	
118	void esp_task_wdt_delete() {
(gdb)
```

As is always the case, non deterministic issues are the trickiest.  The source of the app and the sdkconfig can be found here:

https://github.com/nkolban/esp32-snippets/tree/master/hardware/gps/gps

I plan to study more and run other tests .... but am raising this in the hope that it may be of interest to y'all or else you may say "aha" and know the answer to short circuit my tests.

Spritetm, 2016-11-16:
I caq't reproduce this because your repo doesn't seem to include the minea component you use...

ggsubs, 2016-11-19:
I found the uart interrupt handler overflows on framing errors and comm break: https://github.com/espressif/esp-idf/issues/109 
Very simple fix, no more uart crash since then I excluded these IRQ triggers.

projectgus, 2016-12-22:
@nkolban , do you think it's likely this issue is same as #109 or is it probably something else?

projectgus, 2017-02-16:
I'm going to close this because it seems like the same issue as #109. If the error reoccurs with the current master branch, please reopen.

closed 2017-02-16 by projectgus