espressif/esp-idf #94: vfs and standard file descriptors

jolivepetrus, 2016-11-12:
In vfs file descriptor has two parts: vfs index, and file descriptor into the vfs.

In the case that we want to do a redirection of the standard file descriptors, for example, from the UART to a FAT file system, closing standard file descriptor first, and open the standard file descriptors to a new vfs there are problems using write, read, ... using constants STDIN_FILENO, STDOUT_FILENO and STDERR_FILENO defined in <unistd.h> as:

#define	 STDIN_FILENO	0	/* standard input file descriptor */
#define	STDOUT_FILENO	1	/* standard output file descriptor */
#define	STDERR_FILENO	2	/* standard error file descriptor */

This constants are valid for the first registered vfs, but not for the others.

I solved this temporarily doing this somewhere in the code:

#undef STDIN_FILENO
#define STDIN_FILENO fileno(_GLOBAL_REENT->_stdin)

#undef STDOUT_FILENO
#define STDOUT_FILENO fileno(_GLOBAL_REENT->_stdout)

#undef STDERR_FILENO
#define STDERR_FILENO fileno(_GLOBAL_REENT->_stderr)

igrr, 2016-11-13:
You mean, you first `fclose(stdout)` and then do `stdout=fopen(...)`? This won't work the way you want even on Linux. `stdout` will be redirected, but file descriptor will not stay the same. 

If this i am misunderstanding what you are doing, please post a code snippet to demonstrate.

Spritetm, 2016-11-13:
Actually, under Linux this would work. Closing stdin, stdout and stderr would free up fd 0, 1, 2. Immediately opening a new stdin/out/err would return 0/1/2 for these calls, making the defines work.

igrr, 2016-11-13:
I mean, on Linux libc does not magically assign fd 0 to stdin in all cases. If you fclose stdin first, and then freopen stdout, then stdout will get fd 0 assigned to it. So fd is preserved only for the specific case of a single freopen call. We can make this case work, but at the cost of extra run time overhead.

Spritetm, 2016-11-13:
Obviously, but this technique is used in a few places. Not sure if it's really that common that a workaround is necessary, though.

jolivepetrus, 2016-11-13:
igr, you are right, file descriptors are assigned in the order that you open files, if you open stderr, stdin, stdout file descriptors will be 0, 1, 2, and closing a file implies the reuse of it's file descriptor in the next open.

My case can be solved doing fileno(_GLOBAL_REENT->_stdin) ... in my code.

projectgus, 2016-12-22:
Closing as it seems like this is resolved, but please reopen if I missed something.

closed 2016-12-22 by projectgus