espressif/esp-idf #95: Inclusion of esp_system.h missing bool type definition

nkolban, 2016-11-13:
When I include <esp_system.h> I am finding that the compiler errors out with an indication that it was unable to find a data type called "bool" ... for example:

```
CC esp32_environment.o
In file included from /home/kolban/esp32/esptest/apps/workspace/jerryscript/main/./esp32_environment.c:3:0:
/home/kolban/esp32/esptest/esp-idf/components/esp32/include/esp_system.h:119:1: error: unknown type name 'bool'
 bool system_rtc_mem_read(uint16_t src, void *dst, uint16_t n);
 ^
/home/kolban/esp32/esptest/esp-idf/components/esp32/include/esp_system.h:142:1: error: unknown type name 'bool'
 bool system_rtc_mem_write(uint16_t dst, const void *src, uint16_t n);
```

Spritetm, 2016-11-13:
Seems we're missing something in that header. Fyi, you can fix it for now by doing an #include <stdbool.h> before the esp_system.h include.

nkolban, 2016-11-13:
Oh yes ... prior to the inclusion of the header I have added 

```
#ifndef bool
#define bool        int8_t
#endif
```

So I have a workaround ...

igrr, 2016-11-13:
Oh. Defining bool to uint8_t may break things in very fun ways. Better `#include <stdbool.h>`.

nkolban, 2016-11-13:
Cool!!! ... I got the snippet for using the define for bool from:

https://github.com/espressif/esp-idf/blob/4480ab6c8ce70778df4948e587e712540de237eb/components/esp32/include/xtensa/xos_types.h#L62

However, I'll use stdbool.h from now on.

igrr, 2016-11-24:
Fixed in master.

closed 2016-11-24 by igrr